espressif/esp-who #2: After the program burning, the serial port information has been reset

LilyGO, 2018-10-31:
error message:GPIO isr service is not installed

LilyGO, 2018-10-31:
[image]

XiaochaoGONG, 2018-10-31:
@LilyGO It seems that the allocation of camera frame buffer failed, could you provide more details, like the develop board, camera, or something else?

closed 2018-11-05 by LilyGO

LilyGO, 2018-11-05:
I changed the pin of the camera

LilyGO, 2018-11-05:
#define PWDN_GPIO_NUM     0
#define RESET_GPIO_NUM    15
#define Y2_GPIO_NUM       17
#define Y3_GPIO_NUM       35
#define Y4_GPIO_NUM       34
#define Y5_GPIO_NUM       5
#define Y6_GPIO_NUM       39
#define Y7_GPIO_NUM       18
#define Y8_GPIO_NUM       36
#define Y9_GPIO_NUM       19
#define XCLK_GPIO_NUM     27
#define PCLK_GPIO_NUM     21
#define HREF_GPIO_NUM     26
#define VSYNC_GPIO_NUM    22
#define SIOD_GPIO_NUM     25
#define SIOC_GPIO_NUM     23

XiaochaoGONG, 2018-11-06:
are you using m5camera ? as I know, the Y2_GPIO_NUM should be 32

XiaochaoGONG, 2018-11-15:
Hi @LilyGO, if you are using m5stack, please check the SPI RAM type of chip. Let it be 'Auto detect', it may help.