espressif/esp-who #12: No or broken image sent to browser

xgarb, 2018-12-18:
I'm using an the AI-Thinker ESP32-Cam. I've set up the ESP-IDF environment using MSYS32 and imported this project. For the camera_web_server example everything seems to work except the single image and stream received by the browser don't appear to be valid jpgs.

Similar problem with the ESP32_CAMERA_QR library - https://github.com/donny681/ESP32_CAMERA_QR/issues/7

Terminal output below:

I (0) cpu_start: App cpu up.
I (1257) spiram: SPI SRAM memory test OK
I (1257) heap_init: Initializing. RAM available for dynamic allocation:
I (1257) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1263) heap_init: At 3FFBBA78 len 00024588 (145 KiB): DRAM
I (1270) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1276) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1283) heap_init: At 40094FCC len 0000B034 (44 KiB): IRAM
I (1289) cpu_start: Pro cpu start user code
I (1294) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (195) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (196) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (236) wifi: wifi driver task: 3ffcb4b0, prio:23, stack:3584, core=0
I (236) wifi: wifi firmware version: 328353e
I (236) wifi: config NVS flash: enabled
I (236) wifi: config nano formating: disabled
I (246) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (246) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (276) wifi: Init dynamic tx buffer num: 32
I (276) wifi: Init data frame dynamic rx buffer num: 32
I (276) wifi: Init management frame dynamic rx buffer num: 32
I (276) wifi: Init static tx buffer num: 16
I (286) wifi: Init static rx buffer size: 1600
I (286) wifi: Init static rx buffer num: 10
I (286) wifi: Init dynamic rx buffer num: 32
I (296) camera wifi: wifi_init_softap finished.SSID:ESP32-Camera password:
I (316) camera wifi: wifi_init_sta finished.
I (316) camera wifi: connect to ap SSID:Mueller password:smallhands
I (476) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 2
W (476) phy_init: saving new calibration data because of checksum failure, mode(0)
I (526) wifi: mode : sta (24:0a:c4:29:a1:78) + softAP (24:0a:c4:29:a1:79)
I (526) wifi: Init max length of beacon: 752/752
I (526) wifi: Init max length of beacon: 752/752
I (536) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (546) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (556) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (646) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (646) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (656) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (656) wifi: ap channel adjust o:1,1 n:11,2
I (666) wifi: new:<11,0>, old:<1,0>, ap:<11,2>, sta:<11,0>, prof:1
I (1356) wifi: state: init -> auth (b0)
I (1356) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1356) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1366) wifi: state: auth -> assoc (0)
I (1376) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1376) wifi: state: assoc -> run (10)
I (1386) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1396) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1406) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1416) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1426) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
W (1426) wifi: discard, ba not setup
W (1436) wifi: discard, ba not setup
I (1436) camera: Allocating 2 frame buffers (750 KB total)
I (1486) camera: Allocating 375 KB frame buffer in OnBoard RAM
I (1556) camera: Allocating 375 KB frame buffer in OnBoard RAM
I (1956) camera_httpd: Starting web server on port: '80'
I (1956) camera_httpd: Starting stream server on port: '81'
I (2466) wifi: connected with NSA HONEYPOT, channel 11, bssid = 3c:47:11:87:fc:44
I (2466) wifi: pm start, type: 1

W (2506) wifi: discard, ba not setup
W (2506) wifi: discard, ba not setup
W (2506) wifi: discard, ba not setup
W (2506) wifi: discard, ba not setup
W (2506) wifi: discard, ba not setup
W (2516) wifi: discard, ba not setup
W (2516) wifi: discard, ba not setup
I (5236) event: sta ip: 192.168.1.110, mask: 255.255.255.0, gw: 192.168.1.101
I (5236) camera wifi: got ip:192.168.1.110
W (12806) httpd_uri: httpd_uri: URI '/favicon.ico' not found
W (12806) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI doesn't exist
I (14636) camera_httpd: awb = 0
I (23146) camera_httpd: awb_gain = 0
I (28366) camera_httpd: framesize = 7

XiaochaoGONG, 2018-12-18:
Hi, I think [this](https://github.com/espressif/esp32-camera/issues/5) will help.
BTW, it seems the uri is not right. You may try `192.168.1.101` from the info in the log

pradyumnac, 2019-01-05:
Just a note:
Url is 192.168.1.110 in the log above
101 is the gateway

xgarb, 2019-01-06:
In the end it was a broken camera. I bought a new one and I have it all working now.

closed 2019-01-06 by xgarb