espressif/esp-who #14: Javascript error with Firefox

TheNitek, 2019-01-01:
Firefox reports "SyntaxError: missing ; before statement" on the camera_web_server examples index page.

XiaochaoGONG, 2019-01-03:
Could you provide more information in detail ? What are the settings ?

TheNitek, 2019-01-03:
It's just a standard Firefox and the index page does not work because Firefox considers the Javascript to be broken. I think it has something to do with "let" vs. "const" in the JS code, but I did not have the chance to dive any deeper yet.

XiaochaoGONG, 2019-01-03:
@me-no-dev Do you have any idea about it ?

me-no-dev, 2019-01-03:
@XiaochaoGONG I will have a look. @TheNitek which version Firefox? something recent?

me-no-dev, 2019-01-03:
The page uses some newer JS techniques :) your "let" vs. "const" comment led me to believe that it might be a version issue :) I will see if we can detect and handle this situation better.