espressif/esp-who #16: esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x8000 for file

searchingforcode, 2019-01-04:
$ make flash
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/isti-pc/esp/esp-who/esp-idf/requirements.txt are satisfied.
Flashing binaries to serial port /COM3 (app at offset 0x10000)...
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x8000 for file: C:/msys32/home/isti-pc/esp/esp-who/examples/single_chip/camera_web_server123/build/partitions.bin
make: *** [/home/isti-pc/esp/esp-who/esp-idf/components/esptool_py/Makefile.projbuild:63: flash] Error 2

Why its overlapping? how can i solve it?

XiaochaoGONG, 2019-01-07:
Hi can you post your `partitions.txt` ? It's probably the partition is not satisfied.

closed 2019-03-07 by XiaochaoGONG

ecolyx, 2019-04-23:
I have this issue, partitions.csv contains:

# Espressif ESP32 Partition Table
# Name,  Type, SubType, Offset,  Size
factory, app,  factory, 0x010000, 3M
nvs,     data, nvs,     0x310000, 16K

>python /home/charlie/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 /home/charlie/esp/esp-who/examples/single_chip/camera_web_server/build/bootloader/bootloader.bin 0x10000 /home/charlie/esp/esp-who/examples/single_chip/camera_web_server/build/camera_web_server.bin 0x8000 /home/charlie/esp/esp-who/examples/single_chip/camera_web_server/build/partitions.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x8000 for file: /home/charlie/esp/esp-who/examples/single_chip/camera_web_server/build/partitions.bin

ecolyx, 2019-04-23:
Do I need to open a new issue?

ecolyx, 2019-04-23:
Turns out changing the bootloader log level <> info/verbose solves/causes the problem.

ivanporro, 2020-02-26:
Similar issue here. Compiling 3.3 with Debug or Verbose generate the overlap error. Using Info is just fine.

dale6998, 2020-07-17:
I have had to fall back to Warning in order to get this partition table on the ESP32 with a Macronix SPI Flash:
```
[0] % idf.py --ccache partition_table                                                                                      
Checking Python dependencies...
Python requirements from /home/dale/esp/esp-idf/requirements.txt are satisfied.
Executing action: partition_table
Running ninja in directory /foo/build
Executing "ninja partition_table"...
[1/1] cd /foo/bar/esp-idf/partition_ta...o "*******************************************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,16K,
otadata,data,ota,0xd000,8K,
phy_init,data,phy,0xf000,4K,
ota_0,app,ota_0,0x10000,1600K,
ota_1,app,ota_1,0x1a0000,1600K,
storage,data,fat,0x330000,528K,
*******************************************************************************

```