espressif/esp-who #18: ESP-EYE V2.0 board

buckybots, 2019-01-07:
Is this available?  Where?

XiaochaoGONG, 2019-01-08:
Hi, we will soon provide the link. Now you can use WRoverKit or others with esp32 chip that can support a camera to do face recognition.

martinberlin, 2019-01-09:
@XiaochaoGONG Will be possible in the future to use a OV2640 Arducam but SPI Camera ?
http://www.arducam.com/arducam-mini-released/ I have a couple of them but they use only 3 GPIOs usubg SPI instead of 10 digital pins ( [Demo project uses an OV5642 but same library](https://github.com/martinberlin/FS32) )

XiaochaoGONG, 2019-01-10:
> @XiaochaoGONG Will be possible in the future to use a OV2640 Arducam but SPI Camera ?
> http://www.arducam.com/arducam-mini-released/ I have a couple of them but they use only 3 GPIOs via SPI to read Fifo instead of 10 digital pins ( [Demo project uses an OV5642 but same library](https://github.com/martinberlin/FS32) )

Yes, it can be possible to use the sensor via SPI. To use the algorithm of face recognition, it doesn't care much about how the image data come. So there is a proper one and we use it. As is said in #19, we are seeking sensors with parallel bus.

martinberlin, 2019-01-10:
Alright then I will first try with WRover when it comes, and then once I have it working do a fork and try to read from an SPI camera. But I don't know if it will be enough speed. If the resolution needed it's only QVGA (320x240) or even less then it may work, but I cannot reach parallel speed transmission. The only benefit I see is that they are less cables to connect if you need to develop something with this.

XiaochaoGONG, 2019-01-11:
The 'Raw' mode for the camera output is super huge for us, without any compression. While 'RAW888' or 'RGB888' is what we need, RGB of 24 bits. We do the forward processing on the RGB type of image

Niek, 2019-01-11:
Would love to test this board as well, especially with the added microphone.

vivian-ng, 2019-01-12:
Hoping to try out this board too. When will it be available?

Niek, 2019-01-21:
There is one available for pre-order at https://item.taobao.com/item.htm?spm=a230r.1.14.1.10567878mS2Ipc&id=586201030146&ns=1&abbucket=11#detail - delivery before 2019-02-12 (end of CNY)

andywang2039, 2019-03-14:
@XiaochaoGONG  Hello,
I heard the OV2640 EOL information from image sensor vendor, then how does Espressif  can continue ESP-EYE product life in the future? Do you have plan to change other popular image sensor?

XiaochaoGONG, 2019-03-14:
> @XiaochaoGONG Hello,
> I heard the OV2640 EOL information from image sensor vendor, then how does Espressif can continue ESP-EYE product life in the future? Do you have plan to change other popular image sensor?

Yes we are working on some other sensors, and will provide the driver and the model soon.

closed 2019-03-27 by XiaochaoGONG

MohamadBay, 2019-05-24:
Hey XiaochaoGONG,
I am trying to get the esp-eye running and I am using the Arduino as an IDE and the camera example code of the esp32 wrover.

So far, all good, esp-eye is connected to the network and shows me that I can stream over 192.168.xxx/ normal way.
Unfortunately, the camera is not able to make face recognition or detection.  I pushed the detection and recognition button But it doesn't work. I can only see my face :)
May you please help me here?