espressif/esp-who #19: Question: PIN definitions for external SPI RAM

martinberlin, 2019-01-09:
Can you please document where is the GPIO definition to use an external SPI RAM ?

It's not possible to try this without an external memory right ? 
I just want to try it with another board that is not ESP-WROVER-KIT.

As a second question I would be interested to know if in a near future we can use OV5642 as a camera that has already [an internal memory](http://www.arducam.com/downloads/shields/ArduCAM_Mini_5MP_Camera_Shield_DS.pdf) ? Or there are no plans to support SPI Cameras ?

me-no-dev, 2019-01-09:
The external SPI RAM that is mentioned is actually a chip that is on some of our modules (ESP32-WROVER) and is not meant to be used with other chips. In reality, you can connect compatible PSRAM as it's shown in the ESP32-WROVER schematic: https://www.espressif.com/sites/default/files/documentation/esp32-wrover_datasheet_en.pdf

The SPI camera that you show already includes a processor and is meant for different applications. We are exploring other image sensors as well, but all are using parallel bus as that gives the higher speeds required.

martinberlin, 2019-01-09:
Ah ok thank you for the technical explanation. Makes sense!   
Although I must say that reading images with low resolution like esp-who is quite fast and not more than a couple of miliseconds (betw. 4-8 ms if my measurements are good at SPI 4Mhz)
I just ordered a ESP32-WROVER and in the meantime I will try to connect an external Ram as described in the datasheet.

closed 2019-01-09 by martinberlin