espressif/esp-who #21: How to convert html to hex array?

InfiniteYuan1, 2019-01-10:
How should I convert an html page to hex array？
eg: [camera_index](https://github.com/espressif/esp-who/blob/master/examples/single_chip/camera_web_server/main/include/camera_index.h)

InfiniteYuan1, 2019-01-10:
Is there a tool available?

me-no-dev, 2019-01-10:
I used this to convert the gzipped html to this array: https://gist.github.com/me-no-dev/f137a950ce6dedb641d427d8db6355d2
Note that the html file was first gzipped (`gzip index.html`) and then the result run through the locally compiled binary from the code above (`filetoarray index.html.gz > camera_index.h`)

closed 2019-01-10 by me-no-dev

chegewara, 2019-01-10:
https://linux.die.net/man/1/xxd

me-no-dev, 2019-01-10:
Open it in your browser, right click and view source :)

InfiniteYuan1, 2019-01-10:
yeal, that is right.But I use PC that don't have wireless network, can you help me for provide the html file?Thank you in advance.

me-no-dev, 2019-01-10:
Wait... what do you mean? How will you test this if you have no way to reach the ESP? Your computer does not have Ethernet either? How you connect to internet?

InfiniteYuan1, 2019-01-10:
I use a desktop computer without a wireless network card, I can only access ESP32 through my mobile phone.
However, it is not convenient to view the source code on the phone and convert it to an html file.
Can you provide your camera_index.html file? I want to convert it once.

me-no-dev, 2019-01-10:
[index.html.gz](https://github.com/espressif/esp-who/files/2744676/index.html.gz)

InfiniteYuan1, 2019-01-10:
@me-no-dev @chegewara Thank you.

chegewara, 2019-01-10:
Yes, and js files and even images. Just you need to have setup separate path for each file.

Blu-Eagle, 2021-04-05:
xxd -I file.gz