espressif/esp-who #22: Cannot boot with AI Thinker ESP32-CAM board

Niek, 2019-01-10:
I'm trying to get the single_chip/camera_web_server sample working with my AI Thinker ESP32-CAM board, but so far no luck. Compile + flashing is OK, but it doesn't boot. ``make monitor`` shows:
```
I (32) boot: ESP-IDF v3.3-dev-32-gfb7ba1baa 2nd stage bootloader
I (32) boot: compile time 15:33:32
I (32) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (43) boot: SPI Mode       : DIO
I (47) boot: SPI Flash Size : 4MB
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 factory          factory app      00 00 00010000 00300000
I (69) boot:  1 nvs              WiFi data        01 02 00310000 00004000
I (77) boot: End of partition table
I (81) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1780cc (1540300) map
I (629) esp_image: segment 1: paddr=0x001880f4 vaddr=0x3ff80000 size=0x00000 (     0) load
I (630) esp_image: segment 2: paddr=0x001880fc vaddr=0x3ff80000 size=0x00000 (     0) load
I (636) esp_image: segment 3: paddr=0x00188104 vaddr=0x3ffb0000 size=0x03544 ( 13636) load
I (650) esp_image: segment 4: paddr=0x0018b650 vaddr=0x3ffb3544 size=0x00000 (     0) load
I (654) esp_image: segment 5: paddr=0x0018b658 vaddr=0x40080000 size=0x00400 (  1024) load
```

I'm missing a lot of output above, not sure why.

me-no-dev, 2019-01-10:
did you run menuconfig to select the AI-Thinker board?

Niek, 2019-01-10:
Yes, here my [sdkconfig](https://github.com/espressif/esp-who/files/2745901/sdkconfig.txt) file - I don't see any issues with that.

me-no-dev, 2019-01-10:
yeah... config looks good. And we had reports that it works with that board. We did get some report of bad boards though... maybe look at it closer see if anything comes to mind

me-no-dev, 2019-01-10:
then that is rather strange. the log shows that you do not even get to execute the firmware. Maybe rebuild and reflash?

xgarb, 2019-01-10:
I've just published a 'from scratch' walk-through. Maybe something will help - https://robotzero.one/esp32-camera-module/

me-no-dev, 2019-01-10:
you say there is no support in Arduino IDE while there actually is full support : https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer

XiaochaoGONG, 2019-01-11:
@Niek It's quite strange, did the program stop at `segment 5` or keep resetting ?

Niek, 2019-01-11:
I fixed my monitor issues by using a common ground with the programmer. Now I can see the exact error message, which is:
```
E (747) spiram: Expected 4096KiB chip but found 8192KiB chip. Bailing out..
E (754) cpu_start: Failed to init external RAM!
```

I fixed it by using ``CONFIG_SPIRAM_TYPE_AUTO=y``. I see this is set in sdkconfig.default as well, so I'm not sure how I ended up with ``CONFIG_SPIRAM_TYPE_ESPPSRAM32=y`` - in any case, issue is resolved 👍

closed 2019-01-11 by Niek

xgarb, 2019-01-11:
> you say there is no support in Arduino IDE while there actually is full support : 

Ah! You are quicker at writing code than I am at writing articles. That Arduino IDE version is much easier to set up. I'll try to think of a project for this and write another install guide.

NikunjChiluvuri, 2020-08-17:
does anyone have a Bootloader for esp32 cam?