espressif/esp-who #24: LED / Flash

just-jason, 2019-01-13:
Is it possible to use the front LED of the Ai-Thinker board?

XiaochaoGONG, 2019-01-14:
Surely you can use the LED with the correct GPIO. You could refer to the example here:
https://github.com/espressif/esp-idf/blob/b173cf28174b17f7df96c298f83c16a00d22e312/examples/get-started/blink/main/blink.c#L20

just-jason, 2019-01-14:
Thanks, l did not explain my idea so well, let me try that question again. Would it be possible to include the control over the LED on the AiThinker board in the camera web control panel. I could imagine that the large front white led could be great for adding illumination for the camera, and a useful addition to the web control panel.
[image]

me-no-dev, 2019-01-14:
I have done some tests with another camera and we can even make it flash while it's taking an image. It will give cue and it will serve as a flash. How about it? :)

just-jason, 2019-01-14:
Sounds great ! The LED is mounted up front with the camera, It is going to be fantastic to make use of it in combo with the camera. I could even imagine the idea of having it coupled to the AEC, just an idea.

just-jason, 2019-01-23:
>           I have done some tests with another camera and we can even make it flash while it's taking an image. It will give cue and it will serve as a flash. How about it? :)

Could you tell me how you implemented this in the sketch?

aguaviva, 2019-05-31:
Any updates? I also have a ESP32-CAM and would like the LED to flash when I take a pic or stream video.

bkeevil, 2019-08-08:
Implemented in #99