espressif/esp-who #27: who did the YUV to jpeg encodding?

9crk, 2019-01-17:
this project is awesome!
Does the ov2640 encode the jpeg or ESP32 did the encoding?

9crk, 2019-01-17:
runs 12fps on 640*480/ 6fps on 1024*768/ 25fps under CIF

me-no-dev, 2019-01-17:
The camera can output JPEG also. The way we currently handle it is to get JPEG from the camera, then turn it into RGB888 for detection, recognition and drawing on the frame and then back to JPEG to be pushed to the browser. We have added all necessary tools to convert from any format to BMP(RGB888) or JPEG. When YUV or RGB565 is used, the data flow is too hight and leaves little time for processing. But JPEG on the other hand allows to run at 2MP JPEG without issues.

me-no-dev, 2019-01-17:
Actually 1600x1200@6FPF, 800x600@12FPS and 400x300@25FPS

9crk, 2019-01-17:
sure it does the fps. but need to put the encode quality to 40+
so all the face recognition stuff was  done after turning JPEG to RGB ? is there any benchmark for JPEG decoding and encoding speed ? 

btw. the boot time turns about 5 seconds, is there any possible to speed up?

9crk, 2019-01-17:
this will be a good work to make a doorbell product!

me-no-dev, 2019-01-17:
if you run the web server demo, you will get the timing of each step in the serial console :)

me-no-dev, 2019-01-17:
BTW the FPS largely depends on the board quality and the network quality. You can do it with quality set to 10 if all is good.

martinberlin, 2019-01-17:
A doorbell that can be open with a photo of you does not pass any security standards

closed 2019-01-18 by 9crk