espressif/esp-who #29: i run the demo `camera_web_server` on my board , but  keeps rebootting

xuhongv, 2019-01-18:
- i run the demo `camera_web_server` on my board , but  keeps rebootting , ..

I (693) esp_image: segment 8: paddr=0x002120e4 vaddr=0x4008452c size=0x13370 ( 78704) load
0x4008452c: DPORT_REG_READ at E:/Espressif/Esp32/esp-idf/home/XuHongYss/esp-who/esp-idf/components/soc/esp32/include/soc/dport_access.h:79
 (inlined by) cache_sram_mmu_set at E:/Espressif/Esp32/esp-idf/home/XuHongYss/esp-who/esp-idf/components/esp32/cache_sram_mmu.c:123

I (719) esp_image: segment 9: paddr=0x0022545c vaddr=0x400c0000 size=0x00000 (     0) load
I (719) esp_image: segment 10: paddr=0x00225464 vaddr=0x50000000 size=0x00000 (     0) load
I (726) esp_image: segment 11: paddr=0x0022546c vaddr=0x50000000 size=0x00000 (     0) load
I (748) boot: Loaded app from partition at offset 0x10000
I (748) boot: Disabling RNG early entropy source...
E (750) cpu_start: Failed to init external RAM!

xuhongv, 2019-01-18:
[image]

XiaochaoGONG, 2019-01-18:
Hi, what kind of board are you using ? It looks like your board doesn't support SPI RAM.

XiaochaoGONG, 2019-01-19:
This project needs PSRAM to support face detection and recognition. While the board you proposed doesn't have it.
If you still want to use it but without facial relative functions, you may try `make menuconfig` then to cancel the selection of PSRAM