espressif/esp-who #30: SPI RAM initialization fails

wellsy57, 2019-01-20:
## Environment

- Development Kit:  ESP-WHO Face Detection and Recognition Libraries    
- Kit version: Latest download https://github.com/espressif/esp-who.git
- Module or chip used: ESP32-CAM WiFi + Bluetooth Camera Module Development Board ESP32 With Camera Module OV2640 
- IDF:  MINGW32
- Build System: Make
- Compiler version: 5.2.0
- Operating System: Windows
- Power Supply: USB

## Problem Description

Hi this my first issue here so sorry if I have not followed the process well enough.

I purchased the Geekcreit® ESP32-CAM WiFi + Bluetooth Camera Module Development Board ESP32 With Camera Module OV2640 recently and after finding not a lot of information with the kit I finally settled on trying to set it up using Espressif Systems (https://github.com/espressif) using information found here (https://robotzero.one/esp32-camera-module/) specificly (https://github.com/espressif/esp-who.git).

It has all gone reasonably smoothly with some initial testing using the examples found here (https://github.com/espressif/esp-idf/tree/master/examples) to check the board is working at all. The board seems to pass all tests including the ram being written and read.

However once the  ESP-WHO Face Detection and Recognition Libraries (/esp-who/examples/single_chip/camera_web_server/) is flashed to the module I constantly get on every boot the ERROR (E (841) spiram: SPI RAM enabled but initialization failed. Bailing out.).

I have checked every setting against information found by searching extensively and after trying all solutions offered I have been able to load the camera webserver but the camera fails to initialise due to the SPI RAM failing to initialize I believe? 

### Expected Behavior

The camera should initialize and be available at the IP address which it is successfully allocated. 

### Actual Behavior

The server is available at the allocated IP but no image or stream is available.

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6732
load:0x40078000,len:11960
load:0x40080400,len:7332
entry 0x40080780
I (28) boot: ESP-IDF v3.3-beta1-223-ga62cbfec9 2nd stage bootloader
I (28) boot: compile time 10:53:34
I (28) boot: Enabling RNG early entropy source...
I (34) qio_mode: Enabling default flash chip QIO
I (40) boot: SPI Speed      : 40MHz
I (44) boot: SPI Mode       : QIO
I (48) boot: SPI Flash Size : 4MB
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 factory          factory app      00 00 00010000 00300000
I (70) boot:  1 nvs              WiFi data        01 02 00310000 00004000
I (78) boot: End of partition table
I (82) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x17ecdc (1567964) map
I (557) esp_image: segment 1: paddr=0x0018ed04 vaddr=0x3ff80000 size=0x00000 (     0) load
I (557) esp_image: segment 2: paddr=0x0018ed0c vaddr=0x3ff80000 size=0x00000 (     0) load
I (563) esp_image: segment 3: paddr=0x0018ed14 vaddr=0x3ffb0000 size=0x012fc (  4860) load
I (574) esp_image: segment 4: paddr=0x00190018 vaddr=0x400d0018 size=0x8b674 (570996) map
0x400d0018: _stext at ??:?

I (751) esp_image: segment 5: paddr=0x0021b694 vaddr=0x3ffb12fc size=0x01e00 (  7680) load
I (754) esp_image: segment 6: paddr=0x0021d49c vaddr=0x3ffb30fc size=0x00000 (     0) load
I (757) esp_image: segment 7: paddr=0x0021d4a4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/Gavin/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (767) esp_image: segment 8: paddr=0x0021d8ac vaddr=0x40080400 size=0x17678 ( 95864) load
[... 66 lines cut ...]
I (1183) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1193) wifi: connected with Bugwire, channel 10, bssid = e0:3f:49:72:b8:e0
I (1243) wifi: pm start, type: 1

I (1243) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
############################################################################
CAMERA FAILS TO INITIALISE
############################################################################
I (1253) camera: Allocating 2 frame buffers (750 KB total)
I (1253) camera: Allocating 375 KB frame buffer in PSRAM
E (1263) camera: Allocating 375 KB frame buffer Failed
E (1263) camera: Failed to allocate frame buffer
E (1273) gpio: gpio_isr_handler_remove(380): GPIO isr service is not installed, call gpio_install_isr_service() first
E (1283) camera: Camera init failed with error 0x101
E (1283) app_camera: Camera init failed with error 0x101
I (1293) camera_httpd: Starting web server on port: '80'
I (1303) camera_httpd: Starting stream server on port: '81'
I (2093) event: sta ip: 192.168.1.158, mask: 255.255.255.0, gw: 192.168.1.1
I (2093) camera wifi: got ip:192.168.1.158

```

## Similar issue but found no solution for mine here

https://github.com/espressif/esp-idf/issues/1984

wellsy57, 2019-01-20:
Hi Ivan....thanks the incredibly quick responce!
Yes that is the board.
The blurb says it has - Built-in 520 KB SRAM, external 4M PSRAM.
How do I get it to work with that?

XiaochaoGONG, 2019-01-20:
Hi, without PSRAM you cannot run face detection functions, but still can explore with the camera. Maybe you need to disable PSRAM via `make menuconfig`

wellsy57, 2019-01-20:
Hi XiaochaoGONG...it seems the problem is that the the system expects SPI RAM while the board has PSRAM. I have revisited the settings and make flash again after taking the check out of SPI RAM and now I get a clean run down past the SPI RAM Error but again it has a new error at camera initialization.
```

I (1215) camera: Allocating 375 KB frame buffer in PSRAM
E (1215) camera: Allocating 375 KB frame buffer Failed
E (1225) camera: Failed to allocate frame buffer
E (1225) gpio: gpio_isr_handler_remove(380): GPIO isr service is not installed, call gpio_install_isr_service() first
E (1235) camera: Camera init failed with error 0x101
E (1245) app_camera: Camera init failed with error 0x101
I (1245) camera_httpd: Starting web server on port: '80'
I (1255) camera_httpd: Starting stream server on port: '81'

```

XiaochaoGONG, 2019-01-20:
You may try to change the default size of the image, from `FRAMESIZE_UXGA` to `FRAMESIZE_QVGA`. That won't request so much space to store images, but as a result, you cannot get some high resolution formats due to the limited memory.
https://github.com/espressif/esp-who/blob/994cb1b835b8261392e72356dc4b7cb963fd5844/examples/single_chip/camera_web_server/main/app_camera.c#L69

wellsy57, 2019-01-20:
I have to do some family stuff now so will return with more feedback in around 3-4 hours. Thanks to you both for your help! Very much appreciated!

Cheers

wellsy57, 2019-01-20:
On deck again and after some more checking of RAM settings I believe this error may be significant advice now? What do you both think?

```
E (1375) camera: Failed to allocate frame buffer
E (1375) gpio: gpio_isr_handler_remove(380): GPIO isr service is not installed, call gpio_install_isr_service() first
```
Or is that just a result of the failure to allocate the frame buffer in PSRAM?

XiaochaoGONG, 2019-01-20:
The error is just due to the allocation. What is your image format and size ? As the log has mentioned, that example needs 375KB to store one image, which is beyond the on board RAM.

wellsy57, 2019-01-20:
So after reading your last post I tried to set the the resolution HQVGA in the webserver to see if it would allow the Camera to initialize. No but here is the log result....

``` 
I (94256) camera_httpd: framesize = 3
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d3975  PS      : 0x00060730  A0      : 0x8012c6b2  A1      : 0x3ffd9f90
0x400d3975: cmd_handler at C:/msys32/home/Gavin/esp/esp-who/examples/single_chip/camera_web_server/main/app_httpd.c:532

A2      : 0x3ffd87d8  A3      : 0x00000000  A4      : 0x00000003  A5      : 0x00000001
A6      : 0x3ffda010  A7      : 0x3ffbcd4c  A8      : 0x800d3964  A9      : 0x3ffd9f80
A10     : 0x00000000  A11     : 0x3f403488  A12     : 0x000000ff  A13     : 0x0000ff00
A14     : 0x00ff0000  A15     : 0x40404040  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000008  LBEG    : 0x40094621  LEND    : 0x40094645  LCOUNT  : 0x800d3962
0x40094621: strcmp at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strcmp.S:586

0x40094645: strcmp at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strcmp.S:604

Backtrace: 0x400d3975:0x3ffd9f90 0x4012c6af:0x3ffda000 0x4012c1b1:0x3ffda040 0x4012c25d:0x3ffda0d0 0x4012d70c:0x3ffda0f0 0x4012cf72:0x3ffda110 0x4012cfe4:0x3ffda150 0x400907c1:0x3ffda170
0x400d3975: cmd_handler at C:/msys32/home/Gavin/esp/esp-who/examples/single_chip/camera_web_server/main/app_httpd.c:532

0x4012c6af: httpd_uri at C:/msys32/home/Gavin/esp/esp-idf/components/esp_http_server/src/httpd_uri.c:311

0x4012c1b1: httpd_parse_req at C:/msys32/home/Gavin/esp/esp-idf/components/esp_http_server/src/httpd_parse.c:645

0x4012c25d: httpd_req_new at C:/msys32/home/Gavin/esp/esp-idf/components/esp_http_server/src/httpd_parse.c:645

0x4012d70c: httpd_sess_process at C:/msys32/home/Gavin/esp/esp-idf/components/esp_http_server/src/httpd_sess.c:320

0x4012cf72: httpd_server at C:/msys32/home/Gavin/esp/esp-idf/components/esp_http_server/src/httpd_main.c:113

0x4012cfe4: httpd_thread at C:/msys32/home/Gavin/esp/esp-idf/components/esp_http_server/src/httpd_main.c:113

0x400907c1: vPortTaskWrapper at C:/msys32/home/Gavin/esp/esp-idf/components/freertos/port.c:435

```

Then the device reboots and still the camera fails to initialize. "Allocating 2 frame buffers (750 KB total)" seems to be hard coded (or set somewhere?) in the initialization sequence?

me-no-dev, 2019-01-20:
```cpp
//init with high specs to pre-allocate larger buffers
    config.frame_size = FRAMESIZE_CIF;
    config.jpeg_quality = 12;
    config.fb_count = 1;
```

me-no-dev, 2019-01-20:
this will make it work, but i can not understand why it does not detect your psram if it's there

wellsy57, 2019-01-20:
Yes I agree Me-No-Dev ...Well it does work as I have used the relevant examples to write and read from. Confusing for sure.

So if I save that modification and make flash again you think that will get it working right?

wellsy57, 2019-01-20:
I made the changes to the changes and saved the file.
make flash.....didn't recompile aborted
make menuconfig resaved the settings with no changes

make flash recompiled and flashed the device

make monitor

reboot 

```
I (1215) camera: Allocating 2 frame buffers (750 KB total)
I (1215) camera: Allocating 375 KB frame buffer in PSRAM
E (1225) camera: Allocating 375 KB frame buffer Failed
E (1235) camera: Failed to allocate frame buffer
E (1235) gpio: gpio_isr_handler_remove(380): GPIO isr service is not installed, call gpio_install_isr_service() first
E (1245) camera: Camera init failed with error 0x101
E (1255) app_camera: Camera init failed with error 0x101
```
unfortunately same error?

wellsy57, 2019-01-20:
It seems the "375 KB frame buffer" must be set somewhere else?

chegewara, 2019-01-20:
Check if you have auto-detect or proper SPIRAM selected in menuconfig.

me-no-dev, 2019-01-20:
You are changing these lines, right? https://github.com/espressif/esp-who/blob/master/examples/single_chip/camera_web_server/main/app_camera.c#L69-L71

wellsy57, 2019-01-20:
Present settings: sdkconfig relating SPI RAM 
I will admit I am very confused on how this relates to the fact this board only has PSRAM?
Can anyone explain or offer ideas how it should be set please?
Cheers!

```
#
# SPI RAM config
#
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_IGNORE_NOTFOUND=y
CONFIG_SPIRAM_USE_MEMMAP=
CONFIG_SPIRAM_USE_CAPS_ALLOC=y
CONFIG_SPIRAM_USE_MALLOC=
CONFIG_SPIRAM_TYPE_AUTO=y
CONFIG_SPIRAM_TYPE_ESPPSRAM32=
CONFIG_SPIRAM_TYPE_ESPPSRAM64=
CONFIG_SPIRAM_SIZE=-1
CONFIG_SPIRAM_SPEED_40M=y
CONFIG_SPIRAM_SPEED_80M=
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_CACHE_WORKAROUND=y
CONFIG_SPIRAM_BANKSWITCH_ENABLE=y
CONFIG_SPIRAM_BANKSWITCH_RESERVE=8
CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST=
CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=
CONFIG_PICO_PSRAM_CS_IO=10
CONFIG_MEMMAP_TRACEMEM=
CONFIG_MEMMAP_TRACEMEM_TWOBANKS=
CONFIG_ESP32_TRAX=
CONFIG_TRACEMEM_RESERVE_DRAM=0x0
```

wellsy57, 2019-01-20:
Yes...the whole bottom section is now:
```
    //init with high specs to pre-allocate larger buffers
    config.frame_size = FRAMESIZE_CIF;
    config.jpeg_quality = 12;
    config.fb_count = 1;

    // camera init
    esp_err_t err = esp_camera_init(&config);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Camera init failed with error 0x%x", err);
        return;
    }

    //drop down frame size for higher initial frame rate
    sensor_t * s = esp_camera_sensor_get();
    s->set_framesize(s, FRAMESIZE_QVGA);
```

wellsy57, 2019-01-20:
I have just checked everything again and reflashed after I found that the file I initialled edited and saved was from the wrong location...rooky mistake...sorry.

Now on boot the log reads:

```
I (1326) camera: Allocating 1 frame buffers (14 KB total)
I (1336) camera: Allocating 14 KB frame buffer in OnBoard RAM

```
Now the camera is up and running! Thanks so much for all your help!

Sample of Log while streaming
Stop streaming and select various resolutions and capture a still
```
I (341456) camera_httpd: MJPG: 10660B 69ms (14.5fps), AVG: 79ms (12.7fps), 0+0+0+0=0 0
I (341536) camera_httpd: MJPG: 10638B 78ms (12.8fps), AVG: 79ms (12.7fps), 0+0+0+0=0 0
W (341606) httpd_txrx: httpd_sock_err: error in send : 104
W (341606) httpd_uri: httpd_uri: uri handler execution failed
I (344576) camera_httpd: JPG: 11406B 89ms
I (358606) camera_httpd: framesize = 6
I (362726) camera_httpd: JPG: 14400B 138ms
I (370886) camera_httpd: framesize = 7
I (373446) camera_httpd: JPG: 14400B 89ms
I (380096) camera_httpd: framesize = 8
I (382576) camera_httpd: JPG: 14400B 310ms
I (392696) camera_httpd: framesize = 9
I (396986) camera_httpd: JPG: 14400B 295ms
I (404696) camera_httpd: framesize = 10
I (407326) camera_httpd: JPG: 14400B 196ms
I (423436) camera_httpd: framesize = 5
I (425346) camera_httpd: JPG: 11516B 66ms
I (432636) camera_httpd: framesize = 4
I (435176) camera_httpd: JPG: 7931B 75ms

```
Looks fine below VGA IE: CIF @ VGA it starts to chop off the bottom of the frame.
So the PSRAM chip is faulty?

wellsy57, 2019-01-20:
Hi chegewara....yes it is set to auto but can you look at the settings I posted above and confirm that is as it should be?
I remain confused by all those settings being for SPIRAM when I have PSRAM?

wellsy57, 2019-01-21:
Ok I was able to read from the PSRAM SOP.
Shown as:
IPUS
IPS6404L
1748X

When I searched that I found http://ipusltd.com/Products_en.html
It appears to be a new product they have just started shipping? Apparently it is 64Mbit IoT RAM

So I revisited the menuconfig and found I could choose ESP-PSRAM64 or LY68L6400 as an option?

So I searched that and found:
GENERAL DESCRIPTION
The LY68L6400 is a 64M-bit serial pseudo SRAM
device organized as 8Mx8 bits. It is fabricated using
very high performance, high reliability CMOS
technology.
The LY68L6400 is accessed via a simple Serial
Peripheral Interface(SPI) compatible serial bus.
Additionally, Quad Peripheral Interface(QPI) is
supported if your application needs faster data rates.
This device also supports unlimited reads and writes
to the memory array. 

So it appears this PSRAM can be accessed via SPI right?
So all the settings should be relavent?

Should I just choose the ESP-PSRAM64 or LY68L6400 as the correct option ?

I think I will just try that now anyway and report back.

Still like to hear any other menu selections better suited than mine are at the moment?

wellsy57, 2019-01-21:
No change so I am leaning towards a problem with the board itself at the moment. I measured on the supply to the 3.3V pin and only have 2.7V. Measure on the VCC pin on the RAM and same story 2.7V.
That will be the problem I think? Might have to ask for a replacement I think?

XiaochaoGONG, 2019-01-21:
Could you post a picture of your board ?

wellsy57, 2019-01-21:
I have asked for a replacement from the supplier....may be a good idea to close this issue and I will start a new one if need be later on.

Cheers for everyone’s input...most appreciated!

closed 2019-01-21 by wellsy57

Sanketa54, 2020-10-14:
@wellsy57 :
Hi 
Can u help me with the sdkconfig.h configurations related to psram addition to esp32