espressif/esp-who #32: An error occurs when voice detection and face recognition are run separately.

lewisxhe, 2019-01-22:
When voice detection and face recognition are run separately, an error occurs, and the error log is:

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6700
load:0x40078000,len:10976
load:0x40080400,len:7240
entry 0x40080780
I (29) boot: ESP-IDF v3.3-dev-32-gfb7ba1baa-dirty 2nd stage bootloader
I (29) boot: compile time 11:27:43
I (39) boot: Enabling RNG early entropy source...
I (39) qio_mode: Enabling QIO for flash chip WinBond
I (41) boot: SPI Speed      : 80MHz
I (45) boot: SPI Mode       : QIO
I (49) boot: SPI Flash Size : 4MB
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 factory          factory app      00 00 00010000 00300000
I (72) boot:  1 nvs              WiFi data        01 02 00310000 00004000
I (79) boot:  2 fr               unknown          20 20 00320000 00020000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1c20ec (1843436) map
I (587) esp_image: segment 1: paddr=0x001d2114 vaddr=0x3ff80000 size=0x00000 (     0) load
I (587) esp_image: segment 2: paddr=0x001d211c vaddr=0x3ff80000 size=0x00000 (     0) load
I (594) esp_image: segment 3: paddr=0x001d2124 vaddr=0x3ffb0000 size=0x03b10 ( 15120) load
I (607) esp_image: segment 4: paddr=0x001d5c3c vaddr=0x3ffb3b10 size=0x00000 (     0) load
I (612) esp_image: segment 5: paddr=0x001d5c44 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at D:/espressif/esp-who-new/esp-who/esp-idf/components/freertos/xtensa_vectors.S:1685

I (621) esp_image: segment 6: paddr=0x001d604c vaddr=0x40080400 size=0x09fc4 ( 40900) load
I (643) esp_image: segment 7: paddr=0x001e0018 vaddr=0x400d0018 size=0xa86c4 (689860) map
0x400d0018: _stext at ??:?

I (825) esp_image: segment 8: paddr=0x002886e4 vaddr=0x4008a3c4 size=0x11220 ( 70176) load
0x4008a3c4: lmacMSDUAged at ??:?

I (849) esp_image: segment 9: paddr=0x0029990c vaddr=0x400c0000 size=0x00000 (     0) load
I (849) esp_image: segment 10: paddr=0x00299914 vaddr=0x50000000 size=0x00000 (     0) load
I (856) esp_image: segment 11: paddr=0x0029991c vaddr=0x50000000 size=0x00000 (     0) load
I (880) boot: Loaded app from partition at offset 0x10000
I (880) boot: Disabling RNG early entropy source...
I (882) spiram: Found 32MBit SPI RAM device
I (886) spiram: SPI RAM mode: flash 80m sram 80m
I (891) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (898) cpu_start: Pro cpu up.
I (902) cpu_start: Starting app cpu, entry point is 0x40081520
0x40081520: call_start_cpu1 at D:/espressif/esp-who-new/esp-who/esp-idf/components/esp32/cpu_start.c:245

I (894) cpu_start: App cpu up.
I (1397) spiram: SPI SRAM memory test OK
I (1397) heap_init: Initializing. RAM available for dynamic allocation:
I (1397) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1404) heap_init: At 3FFCD828 len 000127D8 (73 KiB): DRAM
I (1410) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1416) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1423) heap_init: At 4009B5E4 len 00004A1C (18 KiB): IRAM
I (1429) cpu_start: Pro cpu start user code
I (1434) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (112) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (113) esp-eye: Version 0.9.0
I (117) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (148) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[... 78 lines cut ...]

The problem is located in the following function: Since this part is not open source, you can't continue to view it, The comment section will run normally, if it opens, the above error will occur.

```
void nnTask(void *arg)
{
    int audio_chunksize = model->get_samp_chunksize(model_data);
    int16_t *buffer = malloc(audio_chunksize * sizeof(int16_t));
    assert(buffer);

    while (1) {

        xQueueReceive(sndQueue, buffer, portMAX_DELAY);

        // int r = model->detect(model_data, buffer);
        // if (r) {
        //     event_wakeup_detected(r);
        // }

    }

    free(buffer);
    vTaskDelete(NULL);
}
```

lewisxhe, 2019-01-22:
Explain that I did not use the web page to view, but when the machine is started, it will run the detection face directly and will be displayed on the display.

XiaochaoGONG, 2019-01-22:
So..what board are you using ? Without a digital microphone or similar component, you cannot get the audio source, then it will cause some problems.
The example `recognition_solution` should be with [ESP-EYE](https://github.com/espressif/esp-who/blob/master/docs/en/get-started/ESP-EYE_Getting_Started_Guide.md).
For the scenario of face recognition, you need to firstly enroll your face id. Otherwise without any reference of face id, the system will keep going face detection. Surely you can modify the procedure as you want, what I've said is about the example we proposed.

lewisxhe, 2019-01-22:
> So..what board are you using ? Without a digital microphone or similar component, you cannot get the audio source, then it will cause some problems.
> The example `recognition_solution` should be with [...]

Please see( youtube)[https://www.youtube.com/watch?v=vVHurYwPiqI&lc=z23rtjzztp2ssd4hh04t1aokgcwbisqguoy1uqfk0gvabk0h00410] ，Microphone everything works fine .

XiaochaoGONG, 2019-01-23:
I think whatever you have to put in the sample array there should be in the format of 480 integer 16 bits.
https://github.com/espressif/esp-who/blob/994cb1b835b8261392e72356dc4b7cb963fd5844/examples/single_chip/recognition_solution/main/app_speech_recsrc.c#L59-L66

So it may be the data format when you grabbing from I2S interface.

lewisxhe, 2019-01-28:
I didn't change the data format, or I got it in the sample program.

XiaochaoGONG, 2019-01-28:
I mean with your own microphone, you may have to change your data format.

lewisxhe, 2019-01-28:
> I mean with your own microphone, you may have to change your data format.

I mean the microphone used is the same as the ESP-EYE board.

XiaochaoGONG, 2019-01-28:
Please try this patch for the I2S configuration

```
--- a/examples/single_chip/recognition_solution/main/app_speech_recsrc.c
+++ b/examples/single_chip/recognition_solution/main/app_speech_recsrc.c
@@ -20,7 +20,7 @@ static void i2s_init(void)
     i2s_config_t i2s_config = {
         .mode = I2S_MODE_MASTER | I2S_MODE_RX,//the mode must be set according to DSP configuration
         .sample_rate = 16000,                           //must be the same as DSP configuration
-        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,   //must be the same as DSP configuration
+        .channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT,   //must be the same as DSP configuration
         .bits_per_sample = 32,                          //must be the same as DSP configuration
         .communication_format = I2S_COMM_FORMAT_I2S,
         .dma_buf_count = 3,
@@ -43,9 +43,10 @@ void recsrcTask(void *arg)
     i2s_init();
 
     src_cfg_t *cfg=(src_cfg_t*)arg;
-    size_t samp_len = cfg->item_size*2*sizeof(int)/sizeof(int16_t);
+    int buffer_size = cfg->item_size * (sizeof(int32_t)/sizeof(int16_t));
 
-    int *samp=malloc(samp_len);
+    int32_t *samp=malloc(buffer_size);
+    int16_t *samp_out = malloc(cfg->item_size);
 
     size_t read_len = 0;
 
@@ -56,16 +57,15 @@ void recsrcTask(void *arg)
             continue;
         }
 
-        i2s_read(1, samp, samp_len, &read_len, portMAX_DELAY);
-        for (int x=0; x<cfg->item_size/4; x++) {
-            int s1 = ((samp[x * 4] + samp[x * 4 + 1]) >> 13) & 0x0000FFFF;
-            int s2 = ((samp[x * 4 + 2] + samp[x * 4 + 3]) << 3) & 0xFFFF0000;
-            samp[x] = s1 | s2;
+        i2s_read(1, samp, buffer_size, &read_len, portMAX_DELAY);
+        for (int x=0; x<cfg->item_size/2; x++) {
+            samp_out[x] = (int16_t)(samp[x] >> 16) * 8;
         }
 
-        xQueueSend(*cfg->queue, samp, portMAX_DELAY);
+        xQueueSend(*cfg->queue, samp_out, portMAX_DELAY);
     }
 
+
     vTaskDelete(NULL);
 }
```

lewisxhe, 2019-01-29:
> Please try this patch for the I2S configuration
>
> ```
> --- a/examples/single_chip/recognition_solution/main/app_speech_recsrc.c
> +++ b/examples/single_chip/recognition_solution/main/app_speech_recsrc.c [...]
I tried to use this patch, it still reported the same error, I had to disable the microphone

XiaochaoGONG, 2019-01-31:
Could you check the pins to the I2S ? Normally it could work without problems...

lewisxhe, 2019-02-04:
I am pretty sure that there is no problem with the hardware connection, I have uploaded the code to the repository, you can view it here.[](https://github.com/lewisxhe/esp32-camera-screen)

XiaochaoGONG, 2019-02-11:
Hi @lewisxhe , could you take a look at the value of `samp_len` and `read_len`
https://github.com/espressif/esp-who/blob/4dba9c41e9a9304891a69b0ea7ebb36aa73f1eed/examples/single_chip/recognition_solution/main/app_speech_recsrc.c#L45-L64

closed 2019-02-23 by lewisxhe