espressif/esp-who #34: No /dev/cu.* found

cc13com, 2019-02-04:
Hello,

I ordered the following board and work on Mac OS 10.14.1:

_TTGO T-Camera ESP32 WROVER & PSRAM Camera Module ESP32-WROVER-B OV2640 Camera Module 0.96 OLED_

When I try `ls /dev/cu.*` the following is showing:

`
/dev/cu.Bluetooth-Incoming-Port
/dev/cu.MALS
/dev/cu.SOC
`

The same after plugging in the board via usb-kabel and usb-to-usb-c-converter-box.

Any hints what's wrong and no serial port from the board is showing?

Best,
cc13

lewisxhe, 2019-02-04:
Do you have a cp210x driver installed?

martinberlin, 2019-02-05:
Keep in mind also that this board has another GPIOs connected to the camera. So change the example ones to this ones otherwise Camera communication will not work:
https://github.com/lewisxhe/esp32-camera-bme280/blob/master/esp32-camera-bme280.ino

cc13com, 2019-02-05:
> Do you have a cp210x driver installed?

After installation I see `/dev/cu.SLAB_USBtoUART` and set that in `make menuconfig`.

When I start compiling with `make flash` I get the following output.

`
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
AR build/bootloader/micro-ecc/libmicro-ecc.a
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/gpio_periph.o
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/rtc_periph.o
CC build/bootloader/soc/esp32/rtc_pm.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/soc_memory_layout.o
CC build/bootloader/soc/esp32/spi_periph.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
LD build/bootloader/bootloader.elf
Traceback (most recent call last):
  File "/Users/cc13/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 35, in <module>
    import serial.tools.list_ports as list_ports
ImportError: No module named serial.tools.list_ports
make[1]: *** [/Users/cc13/esp/hello_world/build/bootloader/bootloader.bin] Error 1
make: *** [/Users/cc13/esp/hello_world/build/bootloader/bootloader.bin] Error 2
`

Could you help please again.

cc13com, 2019-02-05:
> Keep in mind also that this board has another GPIOs connected to the camera. So change the example ones to this ones otherwise Camera communication will not work:
> https://github.com/lewisxhe/esp32-camera-bme280/blob/master/esp32-camera-bme280.ino

Thanks for this but at the moment I don't know where exactly I have to add/change that. First I will install the hello_world example. ;-)

lewisxhe, 2019-02-05:
You need to install the python serial module using pip and install it in the terminal using the `pip install pyserial` command.

cc13com, 2019-02-05:
> You need to install the python serial module using pip and install it in the terminal using the `pip install pyserial` command.

`Requirement already satisfied: pyserial in /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages (3.4)`

lewisxhe, 2019-02-05:
The red led is the power indicator, the display is dark, please check the code, the board driver pins are 21, 22, the schematic can be found in my repository.

closed 2019-03-27 by XiaochaoGONG