espressif/esp-who #35: how to modify fb.buffer

hughhugh, 2019-02-12:
Hi, I am trying to modify the frame buffer get from the esp32 camera, and then send it to the http server, but I  failed this time. Are there someone know how to solve this problem. Thank you.
The code is listed as the following:
`

//  // ToDo 
//    image = fb->buf;
//    pixels = stbi_load_from_memory(fb->buf, fb->len, &width, &height, &bpp, n_channel);
//
//    if (pixels == NULL) {
//   		ESP_LOGI(TAG,"stbi failure: %s\r\n", stbi_failure_reason());
//     } else {
//
//   	    // added 2019.2.11
//   	    w = width;
//   	    n = n_channel;
//   	    h = height;
//
//   	    for (int dx = 40; dx < 90; ++dx)
//   	    {
//   	    	pixels[n * w * 40 + dx * n + 0] = 255;
//   	    	pixels[n * w * 40 + dx * n + 1] = 0;
//   	    	pixels[n * w * 40 + dx * n + 2] = 0;
//   	    	pixels[n * w * 40 + dx * n + 3] = 255;
//   	    }
//
//   	    for (int dx = 40; dx < 90; ++dx)
//   	    {
//   	    	pixels[n * w * 90 + dx * n + 0] = 255;
//   	    	pixels[n * w * 90 + dx * n + 1] = 0;
//   	    	pixels[n * w * 90 + dx * n + 2] = 0;
//   	    	pixels[n * w * 90 + dx * n + 3] = 255;
//   	    }
//   	    size_t buf_size = strlen((const char *)image);
//   		memcpy(image, pixels, buf_size);
//   		stbi_image_free(pixels);
//     }
//    size_t buf_size = strlen((const char *)image);
//    res = httpd_resp_send(req, (const char *)image, buf_size);`

XiaochaoGONG, 2019-02-14:
Hi, did you get the right pixels or stbi load fails ?
Just a reminder that the fb->buf is in the jpeg format.

hughhugh, 2019-02-17:
I think its right, but I fail to send it out via http server. I have done the same during the QR recognition. It works. But I fail this time . Or do u know some way I can modify the image and then send it to the server. Thank you.

XiaochaoGONG, 2019-02-22:
Hi, you can send the image data with pixels-format like RGB, but the http header type should be modified. 
I suggest to use jpeg format. You can encoder the image using 
https://github.com/espressif/esp32-camera/blob/165a47fe6a7c76daf23fa40512ae4a90a994e938/conversions/include/img_converters.h#L70

hughhugh, 2019-02-23:
Hi,  I  read the codes in esp-who, and they are using the way you show for me. It works.  `httpd_send_chunk()` works. Thanks very much.

hughhugh, 2019-02-25:
@XiaochaoGONG  Hi, I notice that in the project of esp-face,  `libdl_lib.a` is used in the `lib`. Where can I get more introduction about the `lib` library. Thank you.

XiaochaoGONG, 2019-03-07:
We are going to release the library in the future. You can refer the methods of the `lib` from the head files in `lib/include`.

hughhugh, 2019-03-19:
@XiaochaoGONG 您好, 我使用esp32 做图像处理, 尤其是使用了lib/include来进行图像处理后,图像处理好慢,大概需要670ms,请问,esp32里面可以不可以使用dsp加速或者别的图像来加快处理的速度.谢谢!

XiaochaoGONG, 2019-03-19:
> @XiaochaoGONG 您好, 我使用esp32 做图像处理, 尤其是使用了lib/include来进行图像处理后,图像处理好慢,大概需要670ms,请问,esp32里面可以不可以使用dsp加速或者别的图像来加快处理的速度.谢谢!

你好，能说下具体是哪个处理函数吗，基于esp32可以做一些汇编优化。加速模块会在之后的芯片推出，不是目前的esp32。

hughhugh, 2019-03-19:
你好, 我在jpg_stream_httpd_handler{}函数里面做了一些图像运算,使用了
```
                image_matrix = dl_matrix3du_alloc(1, fb->width, fb->height, 3);
                corp_image = dl_matrix3du_alloc(1, 320, 240, 3);
                dst_image = dl_matrix3du_alloc(1, 32, 32, 3);
```
使用了dl_matrix3du_t这个类型进行图像的分割与裁剪,裁剪后的图像发送到httpd.
一个while循环处理流程比较慢.
另外可以采用的汇编优化有哪些,可以使用esp32进行速度优化,0核心时dsp核,对么?谢谢!

hughhugh, 2019-03-19:
```                image_matrix = dl_matrix3du_alloc(1, fb->width, fb->height, 3);                corp_image = dl_matrix3du_alloc(1, 320, 240, 3);                dst_image = dl_matrix3du_alloc(1, 32, 32, 3);```

hughhugh, 2019-03-19:
@XiaochaoGONG 
你好,我们将图像处理的部分封装成一个函数,并使用xTaskCreatePinnedToCore函数绑定到0核和1核上,计算所花费的时间没有变化.不知道怎么可以提高图像处理的速度?谢谢

XiaochaoGONG, 2019-03-19:
参考一下 https://github.com/espressif/esp-dsp，里面的汇编实现

closed 2019-04-22 by XiaochaoGONG