espressif/esp-who #36: ESP32 development board

tiklez107, 2019-02-21:
Does this run with the esp32 development board located here? [https://www.amazon.com/HiLetgo-ESP-WROOM-32-Development-Microcontroller-Integrated/dp/B0718T232Z/ref=sr_1_2_sspa?s=pc&ie=UTF8&qid=1550726606&sr=1-2-spons&keywords=esp32&psc=1](url)
If so, is there any diagrams or specific code for it?

XiaochaoGONG, 2019-02-21:
It is a basic dev board without camera sensor, which cannot run our camera relative programs. But there are still some examples you can refer based on this board: https://github.com/espressif/esp-idf/tree/master/examples

tiklez107, 2019-02-21:
Even if you buy the OV2640 camera module separately?

XiaochaoGONG, 2019-02-21:
Hi, this example with OV2640 is a demo to show face recognition, we also use some other cameras as an option.

greg-ware, 2019-02-21:
Price-wise, you will find much better deals for an ESP-32 board with camera. I'm using M5 CAM which comes for the same price.

tiklez107, 2019-02-21:
The problem is that I need to make my own pcb for a project. I can't use pre-existing boards, only for development purposes. Can I use the ESP-WROVER-B microchip with this?

me-no-dev, 2019-02-21:
you should use wrover for this. wroom does not have psram

tiklez107, 2019-02-21:
> you should use wrover for this. wroom does not have psram

Are there any examples for using the wrover chip? Even to just use the wrover chip with the camera?

me-no-dev, 2019-02-22:
All examples will work, given that you adjust the pinout to whatever you decide it to be.

closed 2019-04-22 by XiaochaoGONG