espressif/esp-who #37: how to optimize the bootup time?

9crk, 2019-02-28:
as I can measure.
 when using webserver demo of ESP-who, app_main was executed about 1S after the reset .
then initial the OV2640 takes about 500ms.
totally takes about 1-2S to get the picture.
how can I reduce this time?

me-no-dev, 2019-02-28:
i am not sure that you can. There are things that need to happen before you are able to grab the first frame. Also you need to go through a few frames in order for the white balance and exposure to get calibrated for your environment

9crk, 2019-02-28:
> i am not sure that you can. There are things that need to happen before you are able to grab the first frame. Also you need to go through a few frames in order for the white balance and exposure to get calibrated for your environment

pls say something optimistic...am I going to quit my project?

9crk, 2019-02-28:
I'm going to disable the AE and AB, so the time comes to be stable under 2S. what can I do next to reduce the time?

me-no-dev, 2019-02-28:
what is your target time? what are you really trying to accomplish?

9crk, 2019-02-28:
> what is your target time? what are you really trying to accomplish?

I want to take a photo immediatlly after a PIR trigger.
this trigger could wake up the ESP32 from low-energy-consumption mode. or from a totally shutdown stat.

hope it could be less than 1S.

9crk, 2019-02-28:
the system have a little battery and hope to use for a long time.

me-no-dev, 2019-02-28:
if you want to capture that fast, you should not put the esp to deep sleep. I understand that you want to conserve the most battery and capture real fast, but technological time is not really something that you can change. When the camera boots, it needs to be reset, wait for the camera to be ready, setup all of it's registers and start acquisition. I doubt that it's possible to do all of that in under a second.
BTW the camera itself also uses quite a bit of power, so long operation on small battery... hard to do

9crk, 2019-03-01:
what if I init the camera in bootloader, so the 1s time was used to stablize the AE and AWB. 
then I can make it about 1S to capture the first usable frame.

does the bootloader opensourced ?

me-no-dev, 2019-03-01:
the bootloader is open and in the components/bootloader inside IDF. You need to setup the camera and reset it before that so time will not really differ.

closed 2019-03-14 by 9crk