espressif/esp-who #40: ESP-EYE v2.1 keeps rebooting

Niek, 2019-03-05:
I bought an ESP-EYE v2.1 from Taobao. After flashing examples/single_chip/camera_web_server on it, I can boot up the ESP-EYE and I see it connects to the WiFi network and gets an IP address.

However, when accessing the IP address in my browser, the ESP-EYE crashes. The serial log is:

```bash
[0;32mI (2765) event: sta ip: 10.37.254.43, mask: 255.255.254.0, gw: 10.37.255.254[0m
[0;32mI (2765) camera wifi: got ip:10.37.254.43[0m
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d67e4  PS      : 0x00060130  A0      : 0x80126bfa  A1      : 0x3ffcb8a0  
A2      : 0x3ffca1ac  A3      : 0x3ffb42e9  A4      : 0x3ffb42e8  A5      : 0x00000000  
A6      : 0x3ffcbc40  A7      : 0x00000001  A8      : 0x800d67ca  A9      : 0x3ffcb880  
A10     : 0x00000000  A11     : 0x3ffca1ba  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000c  LBEG    : 0x400972b5  LEND    : 0x400972c5  LCOUNT  : 0xfffffffd  

Backtrace: 0x400d67e4:0x3ffcb8a0 0x40126bf7:0x3ffcb8c0 0x40125e81:0x3ffcb8f0 0x40125ef2:0x3ffcb980 0x40126238:0x3ffcb9a0 0x40125358:0x3ffcb9c0 0x401253cc:0x3ffcba00 0x4008eeb1:0x3ffcba20

Rebooting...
```

Any idea what's happening here?

XiaochaoGONG, 2019-03-07:
Hi, Niek, did you change the definition Pins for ESP-EYE? If so, please propose more information.

Niek, 2019-03-07:
Wow, I feel stupid about this... I thought the defconfig had the ESP-EYE as default board, but apparently it's the ``WROVER-KIT With OV2640 Module``. Changed it and it boots fine - closing this.

closed 2019-03-07 by Niek