espressif/esp-who #42: Image error When use Grayscale format on XCLK 20MHz

InfiniteYuan1, 2019-03-14:
When I use the grayscale format and the XCLK is 20MHz, the image is obviously wrong. But when the XCLK is 10MHz, the image is normal.

[capture-20MHz.zip](https://github.com/espressif/esp-who/files/2964603/capture-20MHz.zip)

me-no-dev, 2019-03-14:
Maybe you meant to post this in the esp32-camera repository? Which camera are you using?

InfiniteYuan1, 2019-03-15:
yeal,I should post this in the esp32-camera repository.
I use ov2640 with Wrove-Kit board.

closed 2019-03-21 by XiaochaoGONG

InfiniteYuan1, 2019-03-22:
I modify the camera_web_server example only.
```
    config.ledc_channel = LEDC_CHANNEL_0;
    config.ledc_timer = LEDC_TIMER_0;
    config.pin_d0 = Y2_GPIO_NUM;
    config.pin_d1 = Y3_GPIO_NUM;
    config.pin_d2 = Y4_GPIO_NUM;
    config.pin_d3 = Y5_GPIO_NUM;
    config.pin_d4 = Y6_GPIO_NUM;
    config.pin_d5 = Y7_GPIO_NUM;
    config.pin_d6 = Y8_GPIO_NUM;
    config.pin_d7 = Y9_GPIO_NUM;
    config.pin_xclk = XCLK_GPIO_NUM;
    config.pin_pclk = PCLK_GPIO_NUM;
    config.pin_vsync = VSYNC_GPIO_NUM;
    config.pin_href = HREF_GPIO_NUM;
    config.pin_sscb_sda = SIOD_GPIO_NUM;
    config.pin_sscb_scl = SIOC_GPIO_NUM;
    config.pin_pwdn = PWDN_GPIO_NUM;
    config.pin_reset = RESET_GPIO_NUM;
    // Only support 10 MHz current. Camera will output bad image when XCLK is 20 MHz.
    config.xclk_freq_hz = 20000000;
    config.pixel_format = CAMERA_PIXEL_FORM;
    config.frame_size = CAMERA_FRAME_SIZE;
    config.jpeg_quality = 10;
    config.fb_count = 1;

    // camera init
    esp_err_t err = esp_camera_init(&config);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Camera init failed with error 0x%x", err);
        return;
    }
```