espressif/esp-who #43: how to support 640*480 picture ?

9crk, 2019-03-15:
609*480*YUV = 600KB space.

so must use SPIRAM, while the DMA transfer the data to SPIRAM have some problem.

I made a experiment with grayscale picture encoding. while the picture output shows like this:
[image]

what might I did wrong?

9crk, 2019-03-16:
the DMA can only operate 320K address, so the VGA (600K size in YIV format) can not be supported right?

me-no-dev, 2019-03-19:
yes. data is too much for the dma+isr+psram to handle, BUT, we have a new sensor coming (OV3660) which can support 800x600 all formats :)

OV2640 can probably be made to work if the PLL inside is properly set, but there is little to no information on the sensor internals to do that. OV2640 is EOL and not supported by OmniVision anymore.

closed 2019-03-20 by 9crk