espressif/esp-who #44: Importing into PlatformIO

fedyfausto, 2019-03-16:
Hello there is some ways to import the project into a platformIO project?

stripu, 2019-04-09:
> Hello there is some ways to import the project into a platformIO project?

Did you find a way? Or is there any solution to importing this into a easy to use IDE?

me-no-dev, 2019-04-09:
project is included in Arduino, so you have the option to use it with Arduino IDE, PlatformIO or any other IDE with Arduino Support

stripu, 2019-04-09:
> project is included in Arduino, so you have the option to use it with Arduino IDE, PlatformIO or any other IDE with Arduino Support

Thanks for your reply! 
I did use the Arduino example to get the ESP32-CAM working, and it's all good, but i see that in one of the examples here, there's functions to enroll to flash and load back on boot:

enroll_to_flash(): Stores the enrolled Face IDs in the flash
read_id_from_flash(): Reads all the enrolled Face IDs stored in the flash
delete_id_in_flash(): Deletes the earliest enrolled Face IDs stored in the flash

I'd really need these in my project, and the arduino example doesn't have the necessary functions in the header file.

stripu, 2019-04-09:
also, i'm very interested in porting the recognition_with_command_line example in my project :)

stripu, 2019-04-09:
Thank you, me-no-dev :) I guess if we don't figure it out on our own, we can't learn 👍

me-no-dev, 2019-04-09:
this is the point of examples, right :)

closed 2019-05-31 by XiaochaoGONG