espressif/esp-who #45: Trying to get ESpp

JanBosNL, 2019-03-19:


JanBosNL, 2019-03-19:
When using the working code from the github repository https://github.com/donny681/ESP32_CAMERA_QR 

I get this when monitoring the ESP32 

rst' (POWERON_RESEUI0x13 (SPI_FP*e1║M!} ==Q
c˥gsip: 0, SPIWP:
clk_drv:0x00)}rv:0x00,d_drv:00,cs0_drv:0x00,hEv:0x00,wp_drv:00
mode:DIO, clo-"2
load:0x38,len:4
load'fff001c,len:64SH+0x40078000:11308
ho 0 tXZ
         room 4
load'0080400,len:67&H0x4008076SHI (30) boot: ESP-IDF v4.0-dev-78-g8fbb63c2a 2nd stage bootloader
I (30) boot: compile time 11:58:46
I (30) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1c450 (115792) map
I (135) esp_image: segment 1: paddr=0x0002c478 vaddr=0x3ffb0000 size=0x02da4 ( 11684) load
I (140) esp_image: segment 2: paddr=0x0002f224 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/HP/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (142) esp_image: segment 3: paddr=0x0002f62c vaddr=0x40080400 size=0x009e4 (  2532) load
I (152) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x71a14 (465428) map
0x400d0018: _stext at ??:?

I (323) esp_image: segment 5: paddr=0x000a1a34 vaddr=0x40080de4 size=0x0f514 ( 62740) load
I (359) boot: Loaded app from partition at offset 0x10000
I (359) boot: Disabling RNG early entropy source...
I (359) cpu_start: Pro cpu up.
I (363) cpu_start: Application information:
I (368) cpu_start: Project name:     esp32-cam-demo
I (373) cpu_start: App version:      964ba57-dirty
I (379) cpu_start: Compile time:     Mar 19 2019 11:59:04
I (385) cpu_start: ELF file SHA256:  93f82ff032a82d68...
I (391) cpu_start: ESP-IDF:          v4.0-dev-78-g8fbb63c2a
I (397) cpu_start: Starting app cpu, entry point is 0x400810d8
0x400810d8: call_start_cpu1 at C:/msys32/home/HP/esp/esp-idf/components/esp32/cpu_start.c:267

I (0) cpu_start: App cpu up.
I (407) heap_init: Initializing. RAM available for dynamic allocation:
I (414) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (420) heap_init: At 3FFB8D28 len 000272D8 (156 KiB): DRAM
I (427) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (433) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (439) heap_init: At 400902F8 len 0000FD08 (63 KiB): IRAM
I (446) cpu_start: Pro cpu start user code
I (16) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (62) I2S: DMA Malloc info, datalen=blocksize=256, dma_buf_count=8
I (62) I2S: PLL_D2: Req RATE: 39062, real rate: 39062.000, BITS: 16, CLKM: 16, BCK: 8, MCLK: 10000000.000, SCLK: 1249984.000000, diva: 64, divb: 0
I (72) camera_xclk: PIN_CTRL before:3ff,39062
I (82) camera_xclk: PIN_CTRL after:7fff
I (4142) camera_demo: Detected OV2640 camera, using JPEG format
I (4642) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (4642) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (4752) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0
I (4752) camera_demo: Connecting to "MY SSID"

JanBosNL, 2019-03-19:
Got it to work for now, After resetting the ESP32 AI-Thinker board 3 times, suddenly she started working... 

The Cam suplied seems to be a OV2640 eventhough the module itself only has the printed code AF2569 on it. Maybe this is somehow helpfull for someone here.

closed 2019-03-19 by JanBosNL

fensoft, 2020-06-21:
for thoses who have the same problem => i got it working by choosing CAMERA_MODEL_AI_THINKER AND changing my serial2usb cable, as mine fail to provide enough power

martin072, 2021-03-28:
I have a ESP32-CAM with an OV2640 V2, but unfortunately I am not able to initialise it.. It gives the above code 0x20002 Init failed. When I remove the cam it fails with a code of 0x20004, so there is a difference. I tried Both the Esp Cam sketch as well as Tasmota ESP32 with cam support. Both give the same error 0x20002. 

Looked up the Pin config form the seller (https://nl.aliexpress.com/item/4000589847400.html?spm=a2g0s.12269583.0.0.441c7528lgxaiB) and it matches the AI_THINKER setup.. 

Also running it from a stable 5v 2A+ power supply with no difference. Tried to reconnect the tiny cam cable a few times, cannot see any issues with that.

Anyone an idea??