espressif/esp-who #46: How to read camera buffer from DMA directly

Herwey, 2019-03-20:
Hi
Now, I want to improve the speed of getting fb. As far as I am concerned, now we use "fb = esp_camera_fb_get(); " to get the buffer. Is it from cpu? Now, I want to get the camera buffer from DMA directly, how should I do?
Thanks in advance !

me-no-dev, 2019-03-20:
I am not sure I understand what you want, but the esp32-camera driver is open source so you are free to thinker with it.

Herwey, 2019-03-20:
I mean, now as I think, the process of getting image data is like this : camera data -> cpu -> fb. I want to do like this : camera data -> DMA -> fb. So that I can free the cpu and more fast.

me-no-dev, 2019-03-20:
no, data goes like this Sensor -> DMA -> FB

Herwey, 2019-03-20:
I find this line(1182) from camera.c : if (!xTaskCreatePinnedToCore(&dma_filter_task, "dma_filter", 4096, NULL, 10, &s_state->dma_filter_task, 1)).  That means the it uses the core 1 , core 1 is cpu, isn't it?

me-no-dev, 2019-03-20:
the DMA is 16bit but the data is 8bit, so there is a thread that extracts the actual bytes from the DMA buffers.

Herwey, 2019-03-20:
emmm... Actually, I don't understand you very well. So, why 
"if (!xTaskCreatePinnedToCore(&dma_filter_task, "dma_filter", 4096, NULL, 10, &s_state->dma_filter_task, 1))"
it uses cpu core 1 not DMA?  Sorry, I am a new beginner :D

me-no-dev, 2019-03-20:
data flow is like this:
- 8bits from sensor are written to the lower byte of a 16bit DMA buffer item
- when enough bytes are written to the DMA buffer, the DMA signals the filter thread that this buffer is full and starts filling the next one
- the filter thread gets that 16bit DMA buffer and extracts the actual 8bit data from the lower byte of each 16bit sample
- when the frame is complete, the filter thread signals that FB is ready and you get it through the API

is this any clearer? DMA is not made for 8bit data, so the thread is necessary to translate between the DMA and the FB

Herwey, 2019-03-20:
感谢您的回复。但我还是不太明白，因为我现在想通过DMA来获取图像buffer，但我从camera.c里面看到这条指令，
if (!xTaskCreatePinnedToCore(&dma_filter_task, "dma_filter", 4096, NULL, 10, &s_state->dma_filter_task, 1))
这条指令不是把任务分配给core 1吗，这不就是通过cpu了吗？所以是不是可以认为这是通过cpu在获取摄像头图像。

XiaochaoGONG, 2019-03-21:
Hi Herwey, 通过DMA是不能直接得到正确的数据的。原因是DMA传输是16bit为单位，而sensor发过来的数据是8bit为单位，每次接收到的8bit会扩展成16bit，所以数据中间是有一个8bit间隔的，所以这里需要一个转换的过程。
另外，请还是用英文回复吧，不全是会中文的。

closed 2019-05-31 by XiaochaoGONG