espressif/esp-who #47: where is the memory assignned by  "dl_matrix3du_alloc",  internal or external ?

Herwey, 2019-03-22:
Hi
Where is the memory assignned by  "dl_matrix3du_alloc",  internal or external ? 
Thanks in advance !

XiaochaoGONG, 2019-03-22:
That depends on the size you allocate. It will firstly try to allocate the internal memory, if it fails, it would turn to allocate the external.

Herwey, 2019-03-22:
How large about the internal memory. Like : image_matrix = dl_matrix3du_alloc(1, 320, 240, 3);  Is internal memory is enough for this commond?

XiaochaoGONG, 2019-03-27:
You can call `heap_caps_get_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL)` to watch how much is left. Approximately, you might have 250KB - 300KB available.

closed 2019-04-22 by XiaochaoGONG

ChrisXenon, 2019-06-19:
I'm new to this - but I hagve looked extensively an failed to find - the defintion of dl_matrix3du_alloc.
Clearly, from contect, it's allocating memory, presumably for the frame buffer. But why not just use the frame buffer, which is presumably, already in memory?

Obvioulsy, I'm confused - there has to be a good reason to do the dl_matrix3du_alloc() - but I don't know what it is.

Also - is there a place where all these functions are documented? I don't mean with example uses, I mean real documentation with text describing what each function does, and what each parameter is.

Thanks

WayneKeenan, 2019-06-20:
https://github.com/espressif/esp-face/blob/2672de6cda2e65686d87604fb81e8382a390eb9f/lib/include/dl_lib_matrix3d.h#L63

WayneKeenan, 2019-06-20:
but, I can not find the source to the function...

me-no-dev, 2019-06-20:
AFAIK it should try internal first and if not able to, allocate in PSRAM. Either that or the other way round. I know this is not all that helpful, but might give you an idea.

ChrisXenon, 2019-06-20:
Thanks guys. At least we are all agreeing now that there may be a documentation deficit.
me-no-dev (thanks) has told me it's to do with face recognition so at least I can now leave it deleted from my code without worrying about it as I don't need the face stuff.