espressif/esp-who #50: ESP-EYE V2.0 fr_flash: No ID Infomation + internal ram fail

xgarb, 2019-03-25:
I'm trying the recognition_solution example with an ESP-EYE v2.0.

I get the following in the monitor:

**E (6691) fr_flash: No ID Infomation**
I (6701) esp-eye: Version 0.9.0 success
I (43571) wifi: n:1 1, o:1 0, ap:1 1, sta:255 255, prof:1
I (43571) wifi: station: f8:63:3f:4e:50:16 join, AID=1, bgn, 40U
I (43611) app_wifi: station:f8:63:3f:4e:50:16 join, AID=1
I (43641) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
I (104301) app_httpserver: Get count 0

**internal ram fail: 1, 19, 13, 32, size: 31616**

The second error is when I bring up the stream IP in the browser. I've tried it with a few versions of the IDF. The camera_web_server example works fine so I imagine it's a config setting.

Ideas?

XiaochaoGONG, 2019-03-26:
Hi the first one is normal because there is no face ID in the flash. It's just a message to info that face recognition cannot be executed until enrolling at least one face ID.

For the second one, that is the commit id in your example ? It's a hint that the internal ram is running out, so it'll take the external ram. In addition, if the external ram also ran out, that would cause a problem.

xgarb, 2019-03-26:
I remembered this morning that I'd had a problem with accessing another ESP32 board in AP mode. My PC (a Intel NUC NUC7I5BNH  i5-7260U) has problems sending or receiving packets. It works fine when the ESP32 is connected to the Wifi.

I've tried with my laptop and the video stream is working fine. I had thought the errors were fatal rather than informational.

closed 2019-03-27 by xgarb