espressif/esp-who #54: Trigger events (pin I/O) with face recognition without web-interface?

jan-gerard, 2019-04-06:
I am looking for examples or documentation of the face recognition API to trigger events (like switching I/O pins of the ESP32-CAM board). Especially without using the web-interface. I would like to only use the web interface to train the recognition, but then just power the ESP32, have it start face recognition automatically, and respond by switching a voltage on one of the IO pins.
Is that possible? And where do I find which commands to use for that?

me-no-dev, 2019-04-06:
Point in the code where face is recognised: https://github.com/espressif/esp-who/blob/master/examples/single_chip/camera_web_server/main/app_httpd.c#L213

Face Recognition Part: https://github.com/espressif/esp-who/blob/master/examples/single_chip/camera_web_server/main/app_httpd.c#L187-L227

Code that runs the capture, turns it into RGB and runs the face detection (which will call the face recognition part above if face is detected): https://github.com/espressif/esp-who/blob/master/examples/single_chip/camera_web_server/main/app_httpd.c#L243-L318

extract the code above, run it in a loop and toggle a pin or whatever you want to do at the point outlined above :)

closed 2019-04-06 by me-no-dev

xgarb, 2019-04-25:
If you looking for this in Arduino environment I wrote a tutorial including storing the enrolled faces for this: https://robotzero.one/esp32-face-door-entry/

just-jason, 2019-04-25:
> If you looking for this in Arduino environment I wrote a tutorial including storing the enrolled faces for this: https://robotzero.one/esp32-face-door-entry/

Brilliant, thanks very much!!