espressif/esp-who #56: Configure as normal wifi cam

krambriw, 2019-04-11:
Hello, nice product the ESP-EYE !!!
Just started using it. Managed to setup everything and flashed it, works fine 
Just have some questions
- is it possible to configure somehow so that the device acts like a "standard" wifi camera and connects to my home network instead of setting up it's own AP?
- is it possible to make the device autostart instead of needing the the "hi, lexin" voice phrase?

Thank you and kind regards, Walter

krambriw, 2019-04-11:
So I found a solution for my needs, using the example camera_web_server. Just remember to change the definition Pins for ESP-EYE via the menuconfig.
Another question, anyone knows how to configure so that the ESP-EYE has a fixed ip instead of getting one via dhcp in my home router?

krambriw, 2019-04-11:
...and
- how to disable the internal AP (ESP32-Camera)
- how to enable face detection/recognition by default at boot

me-no-dev, 2019-04-11:
I think you should look into ESP-IDF, which is the underlying framework. Those are just examples that combine two ESP-IDF components (esp32-camera and esp-face)

closed 2019-04-14 by krambriw