espressif/esp-who #57: Detected camera not supported

oli4gate, 2019-04-13:
I'm using the ESP-EYE v2.1 kit with the OV2640 camera module and the latest master branch.
With make menuconfig I have selected the ESP_EYE Devkit in the 'Camera Pins' section.
The following error occurs : 
E (273) camera: Detected camera not supported.
E (273) camera: Camera probe failed with error 0x20004
E (273) app_camera: Camera init failed with error 0x20004

I have used the ESP-EYE devkit, before. With an earlier esp-who version about three weeks ago and then it worked fine.

me-no-dev, 2019-04-13:
did you select hardware I2C?

oli4gate, 2019-04-13:
Yes, hardware I2C1 for SCCB is selected.

but both : 
OV2640 support and OV3660 support are selected as well

me-no-dev, 2019-04-13:
huh... I hit the same issue. Though the EYE board works fine with ov3660 and all my other boards work fine with ov2640... not sure what is going on, nothing really changed for ov2640... I'll dig some more

me-no-dev, 2019-04-13:
This has something to do with the pins and I2C. I unplugged the sensor while the camera was still in the USB and then plugged again and rebooted and all is good.

oli4gate, 2019-04-13:
what do you mean with the sensor? Do you mean, you've unplugged the flat cable from the ov2640 camera module while the EYE board is still plugged in the USB. Than plugged the flat cable back into the connector, pushed the rst button and all was good? I would like to try to do the same here to see if I have the same result.

me-no-dev, 2019-04-13:
yes, this is exactly what I did :)

oli4gate, 2019-04-13:
I have got it working like this :
make menuconfig -> Component config -> select OV2460 support
deselect OV3660 support ; Deselect use hardware I2C1 support

me-no-dev, 2019-04-13:
I found it! You did good disabling 3660, no need for I2C. Fix on the way :)

me-no-dev, 2019-04-13:
Done :) the fix is in a pull request. Will be merged probably monday :)
Thanks for reporting this!

dheerajdake, 2019-04-13:
@oli4gate 

> With make menuconfig I have selected the ESP_EYE Devkit in the 'Camera Pins' section.

Where is the Camera Pins section?
I have the following configuration set:
make menuconfig -> Component config -> Camera Configuration
[*] OV2640 Support
[ ] OV7725 Support
[ ] OV3660 Support
[ ] Use hardware I2C1 for SCCB

I get the following error when I run examples/single_chip/detection_with_command_line
https://github.com/espressif/esp-who/issues/59 [Log in this issue]

me-no-dev, 2019-04-13:
answered in the other thread

closed 2019-04-22 by XiaochaoGONG

sahil12051994, 2019-12-03:
I'm not able to work it out. Tried unplugging camera cable keeping USB plugged as well.
[image]

[image]

sahil12051994, 2019-12-03:
> I'm not able to work it out. Tried unplugging camera cable keeping USB plugged as well.
> [image]
> 
> [image]

Worked it out by changing the Camera Pin Settings to my Module i.e AI Thinker which was set to some other by default.