espressif/esp-who #59: Is there any camera app example which works with ESP-EYE v2.1 board?

dheerajdake, 2019-04-13:


dheerajdake, 2019-04-13:
I set menuconfig option to OV2640 and I get this log below:

I (0) cpu_start: App cpu up.
I (794) spiram: SPI SRAM memory test OK
I (795) heap_init: Initializing. RAM available for dynamic allocation:
I (795) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (801) heap_init: At 3FFB4378 len 0002BC88 (175 KiB): DRAM
I (807) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (814) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (820) heap_init: At 4008E538 len 00011AC8 (70 KiB): IRAM
I (826) cpu_start: Pro cpu start user code
I (831) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (180) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (181) sccb: pin_sda 26 pin_scl 27

I (181) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (211) sccb: SCCB_Write [ff]=01 failed

E (211) sccb: SCCB_Write [12]=80 failed

I (231) sccb: SCCB_Probe start
E (1491) camera: Camera probe failed with error 0x20001
E (1491) app_camera: Camera init failed with error 0x20001
E (1491) app_process: Camera capture failed
E (1491) app_process: Camera capture failed
E (1501) app_process: Camera capture failed

me-no-dev, 2019-04-13:
for the web server the camera pinout is selected in menuconfig -> camera web server
in the example you can find the pin definitions, save them and have them for future.

closed 2019-04-13 by me-no-dev

dheerajdake, 2019-04-13:
@me-no-dev 
I found the Camera Web Server option for camera_web_server_config example. However, this option is not there for detection_with_command_line example.

I set the configuration to select ESP-EYE board. Saved the config, exited and when I ran 'make flash', I get the following error:

```
$ make flash
CC build/main/app_httpd.o
fatal error: esp_http_server.h: No such file or directory
compilation terminated.

```

me-no-dev, 2019-04-13:
wrong IDF version

me-no-dev, 2019-04-13:
make sure you run `git submodule update --init --recursive` in the root of esp-who

dheerajdake, 2019-04-13:
I just cloned the repository using 
git clone --recursive https://github.com/espressif/esp-who.git 

I'll try and post an update.

dheerajdake, 2019-04-13:
I tried the command you posted from esp-who folder. It didn't generate any output(as expected as I already did this as a part of clone). I still see the error.

me-no-dev, 2019-04-13:
where does IDF_PATH point to on your computer? It should point to esp-who/esp-idf

dheerajdake, 2019-04-14:
IDF_PATH was pointing to my previous project dir(esp-idf). After updating the path the above error got resolved. Thanks!

I get this error now.
Compiler version: 5.2.0
The following Python requirements are not satisfied:
future>=0.15.2
cryptography>=2.1.4
The recommended way to install a packages is via "pacman". Please run "pacman -Ss <package_name>" for searching the package database and if found then "pacman -S mingw-w64-i686-python2-<package_name>" for installing it.
NOTE: You may need to run "pacman -Syu" if your package database is older and run twice if the previous run updated "pacman" itself.
Please read https://github.com/msys2/msys2/wiki/Using-packages for further information about using "pacman"
WARNING: The cryptography package have dependencies on system packages so please make sure you run "pacman -Syu" followed by "pacman -S mingw-w64-i686-python2-cryptography".

I tried a full system upgrade using pacman -Syu followed by cryptography module update. Updates were successful. But I keep getting this error even after upgrading. Any idea how to resolve this?

dheerajdake, 2019-04-14:
Update:
The below command solved all the dependencies.

`C:/msys32/mingw32/bin/python.exe -m pip install --user -r C:/msys32/home/username/esp/esp-who/esp-idf/requirements.txt
`

I was able to flash camera_web_server app without errors. I see that ESP launched a soft AP. Connected to the soft AP, confirmed connection from the log(make monitor).

I tried going to the IP mentioned in the log(192.168.4.2) but I don't see anything on that page.

Chrome says, "Site can't be reached. Connection refused."

What am I missing here?

me-no-dev, 2019-04-14:
you should have connected to 4.1, where did you see 4.2? maybe that was the IP that the AP gave you?

dheerajdake, 2019-04-15:
I saw 4.2 and sometimes 4.3 in the make monitor log. Anyways, I connected to 192.168.4.1 and finally I was able to see the working app. Thanks!.

The device heats up quickly though.

usernamename99, 2019-07-26:
I see the same error and I'm using the same board.  How do you check the IDF_PATH?  I'm using the ESP-IDF Command prompt so I don't know how to check what IDF_PATH is set.  I could never get anything to build from the MSYS2 shell without errors and found the newer way was to us the command line directly.  The interface is different though and all the info I find refers to msys2.  I'm using idf.py build from ESP-IDF Command prompt.  Everything seems ok until I connect and the camera is not recognized.