espressif/esp-who #60: Support for OV7725

IGRajulu, 2019-04-17:
The menuconfig shows option for OV7725. 
1.
Can I connect a OV7725 Camera ? 
2.
What about the Pin Configuration?

IGRajulu, 2019-04-17:
I'm running camera_web_server example. I work fine with the camera supplied as assembled with ESP-EYE. Now, I'm trying to run the board with OV7725.
Steps followed:
1.
I had connected a standard OV7725 Camera (as available in the market).
2.
Corrected the file "app_camera.c" as below. The file is located at "esp-who\examples\single_chip\camera_web_server\main\":
For: 
config.pixel_format = PIXFORMAT_JPEG;
ChangedTo: 
config.pixel_format = PIXFORMAT_GRAYSCALE;
3.
Selected OV7725 option in menuconfig. 
4.
No errors from 'make flash'.
5.
But 'make monitor' fails. 
Can anybody guess where am I going wrong?
Please notice the two **sccb** related "failed" lines in the log below:

I (422) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (422) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (432) sccb: pin_sda 18 pin_scl 23

**### E (442) sccb: SCCB_Write [ff]=01 failed
E (442) sccb: SCCB_Write [12]=80 failed**

I (472) sccb: SCCB_Probe start
I (542) wifi: n:1 1, o:1 0, ap:1 1, sta:1 0, prof:1
I (1292) wifi: state: init -> auth (b0)
I (1302) wifi: state: auth -> assoc (0)
I (1312) wifi: state: assoc -> run (10)
I (1332) wifi: connected with Senorita02, channel 1
I (1342) wifi: pm start, type: 1

I (1602) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1602) gpio: GPIO[37]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1602) gpio: GPIO[38]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1612) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1622) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1632) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1642) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1652) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1662) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1672) gpio: GPIO[27]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1682) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0

I (1692) camera: Allocating 2 frame buffers (3750 KB total)
I (1692) camera: Allocating 1875 KB frame buffer in PSRAM
I (1962) camera: Allocating 1875 KB frame buffer in PSRAM
I (2242) event: sta ip: 192.168.8.4, mask: 255.255.255.0, gw: 192.168.8.1
I (2242) camera wifi: got ip:192.168.8.4

Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060230  A0      : 0x80128694  A1      : 0x3ffbe160
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffbe1d8  A5      : 0x00000002
A6      : 0x0000000a  A7      : 0x00000005  A8      : 0x801285e6  A9      : 0x3ffbe140
A10     : 0x3ffc9e98  A11     : 0x00000000  A12     : 0x001fbff7  A13     : 0x00000000
A14     : 0x00000640  A15     : 0x000004b0  SAR     : 0x00000005  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x40097960  LEND    : 0x4009796b  LCOUNT  : 0x00000000

0x40097960: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/x
tensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.
S:142

0x4009796b: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/x
tensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.
S:152

Backtrace: 0x00000000:0x3ffbe160 0x40128691:0x3ffbe190 0x400d6e15:0x3ffbe1c0 0x400d55e6:0x3ffb
e260 0x400d12a4:0x3ffbe280 0x400912a1:0x3ffbe2a0
0x40128691: esp_camera_init at E:/esp/msys32/home/GovindWho/esp-who/components/esp32-camera/dr
iver/camera.c:1257

0x400d6e15: app_camera_main at E:/esp/msys32/home/GovindWho/esp-who/examples/single_chip/camer
a_web_server/main/app_camera.c:75

0x400d55e6: app_main at E:/esp/msys32/home/GovindWho/esp-who/examples/single_chip/camera_web_s
erver/main/app_main.c:30

0x400d12a4: main_task at E:/esp/esp-idf/components/esp32/cpu_start.c:494

0x400912a1: vPortTaskWrapper at E:/esp/esp-idf/components/freertos/port.c:435

Rebooting...

me-no-dev, 2019-04-17:
ov7725 is phased out. This is an old sensor that does not support many of the required features.
Why are you trying to get it to work if you have 2640?

IGRajulu, 2019-04-17:
Really ?
Somehow, I got the impression that OV7725 is newer (https://blog.hackster.io/using-a-camera-with-the-esp32-9d6994b34a2b) and better (https://www.kickstarter.com/projects/botthoughts/openmv-cam-embedded-machine-vision/posts/1457847) than OV2640. In AliExpress, OV7725 camera modules are thrice costlier than OV2640; which had led me to that impression.
Can you kindly provide links to any information regarding which is better and/or newer please ?

me-no-dev, 2019-04-17:
the newest sensor that we support is OV3660. It is 3MP sensor that supports JPEG, RGB, YUV and Grayscale output (formats that are not JPEG are limited to 800x600 because we do not have enough RAM). I somehow do like the ov2640 the most so far. I notice that 3660 has some WDR issues and OmniVision were not able to fix that either. If you need JPEG, stick to 2640. If you need one of the other formats, you can go 3660.

me-no-dev, 2019-04-17:
7725 does not support compression. it's 1.3MP sensor AFAIK

closed 2019-04-17 by me-no-dev

IGRajulu, 2019-04-18:
Thank you very much, for the clear-cut guidance.

pramod-wick, 2019-11-27:
I went through the specifications of OV7725 and it has far better low light performance than the OV2640. Is there any way to port the esp who code to work with OV7725

me-no-dev, 2019-11-27:
it does work :) nothing to port

dblatt87, 2020-07-28:
Hello

I try to get an OV7725 running on an ESP32-CAM from ai thinker. 
OV7725 is selectet in camera settings, also OV2640 and OV5640.
OV2640 and OV5640 are working well. 
OV7725 not, these are the errors:

E (3410) camera: Detectet camera not Supported.
E (3410) camera: Camera probe failed with error 0x20001
E (3410) app_camera: Camera init failed with error 0x20001

I also tried it with software I2C and selecting only OV7725, no luck.

Does someone know from similar errors? 
Are there cameras with different pin assignments?

Thanks in advance