espressif/esp-who #61: Still image output is required in PGM format

IGRajulu, 2019-04-18:
Hi,
I want still image output in PGM format.
1.
Do I need to use OV3660 or shall I stick to OV2640?
2.
Will the following change to "app_camera.c" is sufficient to yield PGM format output, instead of JPG? (The file is located at "esp-who\examples\single_chip\camera_web_server\main"):
For:
config.pixel_format = PIXFORMAT_JPEG;
ChangedTo:
config.pixel_format = PIXFORMAT_GRAYSCALE;

XiaochaoGONG, 2019-04-19:
You can get the grayscale value with changing the pixel format. You need also change the code in the example and you don't need to decode Jpeg information.

IGRajulu, 2019-04-20:
Thanks a ton !
"...You need also change the code in the example ..."
Can you kindly elaborate on that please?

XiaochaoGONG, 2019-04-22:
The examples use Jpeg by default, so there need to be decoding process. You need to change nothing in the driver, except your own example. You can get the buffer from `fb = esp_camera_fb_get()` and `fb->buf`, then you can do whatever you want : )

closed 2019-05-28 by XiaochaoGONG