espressif/esp-who #63: hardware I2C1 for SCCB for the OV2640 camera sensor

oli4gate, 2019-04-21:
using the ESP-EYE v2.1 kit with the OV2640 sensor

me-no-dev, 2019-04-21:
Are you sure that you are running the latest camera driver submodule? For me ov2640 works fine with either option.

rochus-keller, 2019-05-27:
I can confirm that with EspEye v2.1 kit and "Use Hardware I2C1 for SCCB" enabled the camera is only recognized when the board is first time USB plugged-in and flashed. 

As soon as I reboot the board I see exactly the same errors as reported by @oli4gate. 

I have three EspEye and was able to reproduce the effect. When I disable "Use Hardware I2C1 for SCCB" and reflash all three boards work again.

I used esp-who commit e53a2a861ccf0396cf2e945c7064a4bf7ca88b39.

me-no-dev, 2019-05-27:
did you update the camera submodule? It needs to include this commit: https://github.com/espressif/esp32-camera/commit/bd617234a74fa291e21dae6996a629af7f6ea5e5

bigbrassbed, 2019-08-04:
I saw that there have been recently some changes in the code camera.c to set the slave address to 0x30 for the OV2640. Hence this doesn't come into effect unless in sccb.c at the line is also change
static uint8_t ESP_SLAVE_ADDR   = 0x3C;
In camera.c we can call SCCB_Write(0x30, 0x12, 0x80);  (e.g. in proc camer_probe) without any effect, when ignoring in sccb.c the provided values

bigbrassbed, 2019-08-04:
... this seem in the meanwhile already fixed, so get the latest code :-)

closed 2019-10-25 by XiaochaoGONG