espressif/esp-who #66: Recognise image from server example

Secretmapper, 2019-05-10:
It would be great if we had an example where the face to be recognised was obtained from a remote endpoint/resource as a reference.

Uploading a picture from a remote website (not in esp32) to remotely update the user that board can recognise

me-no-dev, 2019-05-10:
what is stopping you to do it?

Paidalin, 2019-07-22:
I want to get the video stream in python.

URL="http://192.168.4.1/face_stream"
cap = cv2.VideoCapture(URL)
print(cap.isOpened())

But this code return false. So how can I modify my code?

XiaochaoGONG, 2019-10-25:
The API is for http protocol, while you are using opencv which doesn't implement that protocol.
If you want to get the stream from your own side rather than the camera on board, you'd better modify the source code where getting the image, and then call the functions. 
One possible way is to establish a socket that make the esp32 receive the stream of images, and the other side capture and send that stream.

closed 2019-10-25 by XiaochaoGONG