espressif/esp-who #68: Enroll Face with Name in the CameraWebServer Example

xgarb, 2019-05-14:
I'm stuck with the replacing the enroll_face with enroll_face_with_name function.

int8_t enroll_face(face_id_list *l, dl_matrix3du_t *aligned_face);
to
int8_t enroll_face_with_name(face_id_name_list *l, dl_matrix3d_t *new_id,  char *name);

I've tried replacing dl_matrix3du_alloc and dl_matrix3du_t with the correct versions but then align_face() errors.

```
static int run_face_recognition(dl_matrix3du_t *image_matrix, box_array_t *net_boxes){
    dl_matrix3du_t *aligned_face = NULL;
    int matched_id = 0;

    aligned_face = dl_matrix3du_alloc(1, FACE_WIDTH, FACE_HEIGHT, 3);
    if(!aligned_face){
        Serial.println("Could not allocate face recognition buffer");
        return matched_id;
    }
    if (align_face(net_boxes, image_matrix, aligned_face) == ESP_OK){
        if (is_enrolling == 1){    
            // int8_t enroll_face(face_id_list *l, dl_matrix3du_t *aligned_face);
            int8_t left_sample_face = enroll_face(&id_list, aligned_face);
            // int8_t enroll_face_with_name(face_id_name_list *l, dl_matrix3d_t *new_id,  char *name);
 
```
Should I be able to do this?

XiaochaoGONG, 2019-05-15:
Ah, the meaning for `enroll_face` and `enroll_face_with_name` is not same. The former function will extract the feature of the face and add to a list, the latter function will just add the extracted feature to the list. Before you call the `enroll_face_with_name`, you should get the feature. Concretely, please refer to the example https://github.com/espressif/esp-who/tree/master/examples/single_chip/recognition_wechat

xgarb, 2019-05-20:
I've used the code in the wechat example and have this working pretty well...

[image]

The only part that I have a problem with is deleting a single face. I use this JS to call the delete function on the ESP

`const query = `${baseHost}/control?req_person=${person_name}&req_state=face_delete`;
`
Delete function looks like this: 

```
  char person[FACE_ID_SAVE_NUMBER * ENROLL_NAME_LEN] = {0,};
  char state[32] = {0,};

  qbuf_len = httpd_req_get_url_query_len(req) + 1;
  if (qbuf_len > 1) {

    qbuf = (char*)malloc(qbuf_len);
    if (httpd_req_get_url_query_str(req, qbuf, qbuf_len) == ESP_OK) {

      if (httpd_query_key_value(qbuf, "req_person", person, sizeof(person)) == ESP_OK) {

        memcpy(st_name.enroll_name, person, strlen(person) + 1);

      }
      if (httpd_query_key_value(qbuf, "req_state", state, sizeof(state)) == ESP_OK) {
      }
    }
  }

  if (!strcmp(state, "face_delete"))
  {
    // ["aaa","bbb"]
    char delete_name[ENROLL_NAME_LEN * FACE_ID_SAVE_NUMBER];
    uint8_t len = strlen(person) - strlen("%5b%22") - strlen("%22%5d");
    memcpy(delete_name, person + strlen("%5b%22"), len);
    delete_name[len] = 0;
    char *p = delete_name;
    char *q = NULL;

    do
    {
      q = strstr(p, "%22%2C%22");
      if (!q)
        break;
      p[q - p] = 0;
      int8_t left = delete_face_id_in_flash_with_name(&st_face_list, p);
      if (left > 0)
        ESP_LOGW(TAG, "%s ID Delete", p);
      p = q + 9;
      q = strstr(p, "%22%2C%22");
    } while (q);
    int8_t left = delete_face_id_in_flash_with_name(&st_face_list, p);
    if (left > 0)
      ESP_LOGW(TAG, "%s ID Delete", p);

    const char *response;
    response = person;
    httpd_resp_set_type(req, "text/plain");
    httpd_resp_set_hdr(req, "Access-Control-Allow-Origin", "*");
    return httpd_resp_send(req, response, strlen(response));
  }

```
I get a 'Stack smashing protect failure!' error

with this stack result

> Decoding stack results
> 0x40091648: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
> 0x40091879: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
> 0x400e3020: __stack_chk_fail at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/stack_check.c line 36
> 0x400d2a13: cmd_handler(httpd_req_t*) at C:\Users\Dude\AppData\Local\Temp\arduino_build_842614\sketch\app_httpd.cpp line 481
> 0x4012e499: httpd_uri at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp_http_server/src/httpd_uri.c line 218
> 0x4012d715: httpd_parse_req at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp_http_server/src/httpd_parse.c line 527

Any ideas? If this isn't enough code I'll link to the tutorial I'm writing today or tomorrow.

xgarb, 2019-05-26:
In the end just the line

`delete_face_id_in_flash_with_name(&st_face_list, person);`

worked for me

You can see the full tutorial here: https://robotzero.one/esp-who-recognition-with-names/  I have one problem at the moment with it... it always 'recognises' a face even when it's not been enrolled. The WeChat code worked when I hacked it about but I must have an error somewhere in my version.

XiaochaoGONG, 2019-05-28:
Not clear about your code at app_httpd.cpp line 481.
And about the recognition issue, when you put on someone not enrolled, what is the similarity score ?
Some people also said that among certain versions, the recognition system recognizes all as one same person, but I tested the latest version from now on, there isn't this phenomenon. So please try it again.

xgarb, 2019-06-03:
Which line is it? The code only goes up to 343.

The problem I have is it always matches a face even when the face hasn't been enrolled. I tried an older MTMN config and it was the same. It's probably something wrong in my code.

Is there a way I can print the face names and score in the serial monitor?

XiaochaoGONG, 2019-06-04:
#71 might solve the problem. Not only add the declaration, but need to update libraries in esp-face.

> 0x400d2a13: cmd_handler(httpd_req_t*) at C:\Users\Dude\AppData\Local\Temp\arduino_build_842614\sketch\app_httpd.cpp line 481

From the trace, it said something in line 481 was wrong.

And the format of deleting the face id is only adjust the wechat mini program, you can easily use `delete_face_id_in_flash_with_name` with the name that you have well formatted. If you just pass a single string, you don't have to process some punctuation like comma, space, etc.

xgarb, 2019-06-04:
Ah, ok.. I'm using Arduino IDE so I'll wait for the dev version to catch up.

I see why you see that error.. I don't have the app_httpd.cpp file in my project. I put everything in the ino file except for the pin definitions and the html.gz.  I find it easier to see what's happening.

I realised later that the Wechat needed the extra code to remove any unwanted spaces etc as the data entry isn't a simple text field.

Thanks for you help. I'll close this issue.

closed 2019-06-04 by xgarb