espressif/esp-who #69: Running face detection crashes after a while

HelloDB, 2019-05-14:
Examples:`detection_with_command_line`
```py
[env:esp32dev]
; platform = https://github.com/platformio/platform-espressif32.git#feature/stage
; platform = https://github.com/platformio/platform-espressif32.git ; upstream_develop
platform = espressif32 ; Latest stable version 
board = esp32dev
framework = arduino
upload_speed = 921600
board_build.f_cpu = 240000000L
build_flags =
    -DCORE_DEBUG_LEVEL=3
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
board_build.partitions = huge_app.csv
```
Crash after a few seconds or tens of seconds of normal operation.
I saw several errors, but I don't know how to deal with them.
I would be grateful for any help!
```cpp
void task_process (void *arg)
{/*{{{*/
    size_t frame_num = 0;
    dl_matrix3du_t *image_matrix = NULL;
    camera_fb_t *fb = NULL;
    mtmn_config_t mtmn_config = init_config();

    // int face_id = 0; 
    // uint8_t * _jpg_buf = NULL;
    // size_t _jpg_buf_len = 0;
    
    do
    {
        int64_t start_time = esp_timer_get_time();
        fb = esp_camera_fb_get();
        if (!fb)
        {
            ESP_LOGE(TAG, "Camera capture failed");
            continue;
        }
        int64_t fb_get_time = esp_timer_get_time();
        ESP_LOGI(TAG, "Get one frame in %lld ms.", (fb_get_time - start_time) / 1000);

        image_matrix = dl_matrix3du_alloc(1, fb->width, fb->height, 3);
        uint32_t res = fmt2rgb888(fb->buf, fb->len, fb->format, image_matrix->item);
        if (true != res)
        {
            ESP_LOGE(TAG, "fmt2rgb888 failed, fb: %d", fb->len);
            dl_matrix3du_free(image_matrix);
            continue;
        }

        esp_camera_fb_return(fb);

        box_array_t *net_boxes = face_detect(image_matrix, &mtmn_config);
        ESP_LOGI(TAG, "Detection time consumption: %lldms", (esp_timer_get_time() - fb_get_time) / 1000);
   
        if (net_boxes)
        {
            frame_num++;
            ESP_LOGI(TAG, "DETECTED: %d\n", frame_num);
            // draw_face_boxes(image_matrix, net_boxes, face_id); // 标记人脸
            // free(net_boxes->score);
            free(net_boxes->box);
            free(net_boxes->landmark);
            free(net_boxes);
        }

        // if(!fmt2jpg(image_matrix->item, fb->width*fb->height*3, fb->width, fb->height, PIXFORMAT_RGB888, 90, &_jpg_buf, &_jpg_buf_len)){
        //     Serial.println("fmt2jpg failed");
        //     res = ESP_FAIL;
        // }
		dl_matrix3du_free(image_matrix);
        // drawArrayJpeg(_jpg_buf, _jpg_buf_len, 0, 0);
        // _jpg_buf = NULL;

    } while(1);
}/*}}}*/
void app_facenet_main()
{
    xTaskCreatePinnedToCore(task_process, "process", 4 * 1024, NULL, 5, NULL, 1);
}
```
```py
matrix3d item alloc failed.
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e4dab  PS      : 0x00060530  A0      : 0x800e4fa4  A1      : 0x3ffaf350  
A2      : 0x00000000  A3      : 0x3f4563a0  A4      : 0x0000000b  A5      : 0x00000020  
A6      : 0x3ffafd68  A7      : 0x00000f20  A8      : 0x00000000  A9      : 0x3ffaf320  
A10     : 0x3ffafd68  A11     : 0x400883f0  A12     : 0x00000404  A13     : 0x00003c80  
A14     : 0x00000000  A15     : 0x3ffae740  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff8  

Backtrace: 0x400e4dab:0x3ffaf350 0x400e4fa1:0x3ffaf370 0x400e5f55:0x3ffaf3e0 0x400fa435:0x3ffaf420 0x400e2980:0x3ffaf520 0x400e2e53:0x3ffaf5a0 0x400d1cff:0x3ffaf620 0x4008cd19:0x3ffaf6b0

PC: 0x400e4dab: dl_matrix3d_init_bias at /home/gongxiaochao/code/esp/dl_lib_git/components/dl_lib/dl_lib_matrix3d.c line 37
EXCVADDR: 0x00000008

Decoding stack results
0x400e4dab: dl_matrix3d_init_bias at /home/gongxiaochao/code/esp/dl_lib_git/components/dl_lib/dl_lib_matrix3d.c line 37
0x400e4fa1: dl_matrix3d_conv at /home/gongxiaochao/code/esp/dl_lib_git/components/dl_lib/dl_lib_matrix3d.c line 524
0x400e5f55: dl_matrix3d_mobilenet at /home/gongxiaochao/code/esp/dl_lib_git/components/dl_lib/dl_lib_matrix3d.c line 1058
0x400fa435: onet_with_score_verify at /home/gongxiaochao/code/esp/dl_lib_git/components/face_detection/mtmn/mtmn.c line 229
0x400e2980: o_net_forward at /Users/ficeto/Desktop/ESP32/ESP32/public-libs-build/components/esp-face/face_detection/fd_forward.c line 243
0x400e2e53: face_detect at /Users/ficeto/Desktop/ESP32/ESP32/public-libs-build/components/esp-face/face_detection/fd_forward.c line 342
0x4008cd19: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```
```py
[E][to_jpg.cpp:213] fmt2jpg(): JPG buffer malloc failed
fmt2jpg failed
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1f9a  PS      : 0x00060d30  A0      : 0x800d1ffe  A1      : 0x3ffaf4b0  
A2      : 0x3ffc0408  A3      : 0x3ffc051c  A4      : 0x000000fc  A5      : 0x3ffc0516  
A6      : 0x00000000  A7      : 0x00000051  A8      : 0x3ffc051c  A9      : 0x00000000  
A10     : 0x00000001  A11     : 0x3ffc0618  A12     : 0x0000000e  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

Backtrace: 0x400d1f9a:0x3ffaf4b0 0x400d1ffb:0x3ffaf4d0 0x400d2a72:0x3ffaf4f0 0x400d2ae2:0x3ffaf510 0x400d2b72:0x3ffaf530 0x400d2bcd:0x3ffaf560 0x400d3f91:0x3ffaf580 0x400d2044:0x3ffaf5c0 0x400d2129:0x3ffaf5e0 0x400d1c14:0x3ffaf600 0x400d1e1b:0x3ffaf620 0x4008cd19:0x3ffaf6b0

PC: 0x400d1f9a
EXCVADDR: 0x00000000

Decoding stack results
0x4008cd19: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```
[image]

XiaochaoGONG, 2019-05-15:
Hi did you enable the PSRAM option ? It seems the memory was out, but it should not happen if with PSRAM

HelloDB, 2019-05-15:
PSRAM is enabled
```py
build_flags =
    -DBOARD_HAS_PSRAM
```

XiaochaoGONG, 2019-05-28:
Hi, do you still have the problem ?

HelloDB, 2019-05-29:
Yes, will it be a hardware problem? I use my own PCB and the early ESP32-WROVER module. Will the power ripple cause the PSRAM to go wrong?
[image]
[image]

XiaochaoGONG, 2019-05-29:
When it starts booting, you could read a message like "spiram: Adding pool of 4096K of external SPI memory to heap allocator" that indicates you have successfully joined the spiram into heap.

HelloDB, 2019-05-29:
Initialization is normal, print "PsramSzie: 4194252 bytes", he can run for a period of time and can recognize the face, after a period of time, the crash restarts.

XiaochaoGONG, 2019-05-30:
Examples:detection_with_command_line and the current version ?
From the code above, this part `// free(net_boxes->score);` should not be commented.

HelloDB, 2019-05-30:
There is no ‘score’ in the latest stable version of ‘image_util.h -> box_array_t’. After using the staging version, uncomment the score, but the problem still exists.
[image]

HelloDB, 2019-05-30:
Always crashes after detecting 63 (58~63) faces, it really looks like some memory is not released.

XiaochaoGONG, 2019-06-03:
Hi, in the latest version of esp-face, it does have `score`. Did you forget update the submodule with `git submodule update` ? And after that, please use `CMake` to build the system. https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/index.html#

XiaochaoGONG, 2019-07-16:
The bug is fixed. 243d0bb9f71b07c50c5d7c6af33b5400315e8641

closed 2019-07-16 by XiaochaoGONG