espressif/esp-who #70: Esp Eye with arduino

MohamadBay, 2019-05-24:
I am trying to get the esp-eye running and I am using the Arduino as an IDE.

So far, all good, esp-eye is connected to the network and shows me that I can stream over 192.168.1***.
Unfortunately, the camera can not make face detection and recognisation.  I pushed the face recognition and face detection button But it doesn't work.

May you please help me here?

xgarb, 2019-05-24:
Face detection and recognition aren't working in 1.0.2 of the ESP32 board libraries. You might need to roll back to 1.0..1. Tutorial under **Adding or Updating the ESP32 Range in the Arduino IDE** if you are stuck with this: https://robotzero.one/esp32-cam-arduino-ide/

MohamadBay, 2019-05-24:
I rolled back to 1.0.1 version. But there you are not able to init the esp eye.
you have only:
#define CAMERA_MODEL_WROVER_KIT
#define CAMERA_MODEL_M5STACK_PSRAM
#define CAMERA_MODEL_AI_THINKER

NO ESP-EYE

Do you have an idea?

xgarb, 2019-05-24:
The ESP-EYE definitions can be found here: https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/camera_pins.h 

You also need this in void setup()
#if defined(CAMERA_MODEL_ESP_EYE)
  pinMode(13, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
#endif

If you look at the differences in the example you have now and the example at https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/ you should be able to fix it by bringing over the ESP-EYE related code.

MohamadBay, 2019-05-24:
and where should I paste the : #if defined(CAMERA_MODEL_ESP_EYE)
pinMode(13, INPUT_PULLUP);
pinMode(14, INPUT_PULLUP);
#endif

in the void setup() ? on which position?

xgarb, 2019-05-24:
Above
 // camera init

same as here: https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Camera/CameraWebServer/CameraWebServer.ino

MohamadBay, 2019-05-24:
thank you but   Face detection and recognition is not working  
1.0.1 not working
1.0.2 not working
a other member has the same Problem : https://github.com/espressif/arduino-esp32/issues/2817

any idea??

mctosima, 2019-07-11:
So the conclusion is, if i wanted to develop face recognition and send it to AWS, better to use flash from terminal rather than arduino IDE ?