espressif/esp-who #71: It recognizes every face, not just mine !

JB-DX, 2019-05-27:
I trained it to my (male) face, and it does recognize it.
But when holding several magazine pages with different  faces in front of the camera it  thinks it is me, even when it is a female face.  Seems everything with eyes, nose and mouth is enough for acceptance.  Have I missed the purpose, or is it really so bad distinguishing faces?

XiaochaoGONG, 2019-05-28:
There are some problems among certain commits, so please update to the latest version and try again.

JB-DX, 2019-05-28:
I just got the latest commit. The result is still the  same. It accepts the face printed in the magazine as mine. Have you ever tested this situation?

XiaochaoGONG, 2019-05-28:
Yes, with esp-face @ 689baaa , I've no problem with that. What is the submodule commit of your esp-who ?

JB-DX, 2019-05-28:
I explicitly downloaded @ 689baaa again but the result remains the same.
What do you mean with "submodule commit" ?
From esp-who I yesterday downloaded app_httpd.c , only the mtmn_config parameters were modified compared to my older version. I took these new values.

XiaochaoGONG, 2019-05-31:
Hi I notice that when I use Make system to compile the project, the lib in the esp-face does not work as expected. While with CMake to build the project, there is no problem. Before I find the difference, you can try the cmake process.

JB-DX, 2019-05-31:
I tried to set up CMake for Windows, but get errors with idf.py menuconfig. The process is not well thought out. I am wasting a lot of time with it, and have no idea how to fix it. Help is non existing!

After all, why is the make-system responsible for the difference? As long as compiler and linker don't complain the result should be identical. The relevant source files are only a few and one would see compilation errors, right? It would more helpful to find the real cause.

XiaochaoGONG, 2019-06-03:
Hi, please update to esp-face to 9b4c3ee.

JB-DX, 2019-06-04:
That works better now. Magazine images are now treated differently. Also my face is often not recognized immediately, especially when one half is a bit darker (when the light comes from the side).  That slowness I have experienced in an earlier version, so it is a bit familiar.

closed 2019-06-04 by XiaochaoGONG