espressif/esp-who #75: recognition_solution - jpeg encode failed

tatsutaigu, 2019-06-04:
Hi,

I try sample code  `recognition_solution`.
I found error and ESP-EYE rebooted.
How do I check and revise it?

> I (116772) face_recognition:
> Similarity: 0.999166, id: 1
> I (118542) face_recognition:
> Similarity: 0.998936, id: 1
> I (120252) face_recognition:
> Similarity: 0.998795, id: 1
> E (122022) to_bmp: JPG encoder init failed
> E (122022) app_httpserver: fmt2jpg failed

me-no-dev, 2019-06-04:
Are you using the example as-is or did you change something in the code? Error points to you running out of RAM.

XiaochaoGONG, 2019-06-04:
Apart from the encoding failure, the similarity is abnormal. Please update to the latest version to fix the recognition issue.

tatsutaigu, 2019-06-05:
I tried latest version without modification code today.
However, I got 2 type error.
usually, I got **Type 2** error.

**Type 1**
> I (23584) camera: Allocating 2 frame buffers (30 KB total)
> I (23584) camera: Allocating 15 KB frame buffer in OnBoard RAM
> I (23594) camera: Allocating 15 KB frame buffer in OnBoard RAM
> I (23724) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1
> I (23724) esp-eye: Version 0.9.0 success
> I (55614) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
> I (55614) wifi: station: 98:ca:33:2b:5f:58 join, AID=1, bgn, 20
> I (55614) app_wifi: station:98:ca:33:2b:5f:58 join, AID=1
> dhcps: send_offer>>udp_sendto result 0
> I (56784) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
> W (78114) httpd_uri: httpd_uri: URI '/' not found
> W (78114) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI doesn't exist
> I (83624) app_httpserver: Get count 1
> 
> I (93674) face_recognition: ne_ratio  : 2.798128
> I (94044) face_recognition: ne_ratio  : 2.717711
> I (94834) face_recognition: ne_ratio  : 2.413797
> I (97884) face_recognition:
> Similarity: -0.063272, id: -1
> I (98734) face_recognition:
> Similarity: -0.177422, id: -1
> I (104684) face_recognition:
> Similarity: -0.151737, id: -1
> Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
> Core 0 register dump:
> PC      : 0x4015cada  PS      : 0x00060134  A0      : 0x800d1e01  A1      : 0x3ffbeb80
> A2      : 0x00000000  A3      : 0x00000001  A4      : 0x80092e9a  A5      : 0x3ffd3bb0
> A6      : 0x00000003  A7      : 0x00060023  A8      : 0x00060023  A9      : 0x3ffbec6c
> A10     : 0x00000001  A11     : 0x00000001  A12     : 0x80092e9a  A13     : 0x3ffd3bb0
> A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005
> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
> 
> Backtrace: 0x4015cada:0x3ffbeb80 0x400d1dfe:0x3ffbeba0 0x4009225d:0x3ffbebc0 0x40092ca1:0x3ffbebe0
> 
> Core 1 register dump:
> PC      : 0x4008734d  PS      : 0x00050034  A0      : 0x400854a8  A1      : 0x3ffb0be0
> A2      : 0x3ffc33fc  A3      : 0x3f41dd28  A4      : 0x00000001  A5      : 0x40093c54
> A6      : 0x000000fd  A7      : 0x406fa000  A8      : 0x00000001  A9      : 0x000000fd
> A10     : 0x7ff00000  A11     : 0x7fffa000  A12     : 0x00000000  A13     : 0x00000007
> A14     : 0x00000000  A15     : 0x406fa000  SAR     : 0x00000020  EXCCAUSE: 0x00000005
> EXCVADDR: 0x00000000  LBEG    : 0x40002390  LEND    : 0x4000239f  LCOUNT  : 0x00000000
> 
> Backtrace: 0x4008734d:0x3ffb0be0 0x400854a5:0x3ffb0c10 0x4013202c:0x3ffeaa20 0x4012f86e:0x3ffeaad0 0x40130312:0x3ffeab50 0x400d5ccd:0x3ffeabb0 0x4012d477:0x3ffead60 0x4012ebf5:0x3ffeada0 0x4012ecd1:0x3ffeae30 0x4012e45c:0x3ffeae50 0x4012dcbe:0x3ffeae70 0x4012dd30:0x3ffeaeb0 0x40092ca1:0x3ffeaed0
> 
> Rebooting...
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

backtrace is here.

> $ xtensa-esp32-elf-addr2line -e recognition_http.elf -af 0x4008734d:0x3ffb0be0 0x400854a5:0x3ffb0c10 0x4013202c:0x3ffeaa20 0x4012f86e:0x3ffeaad0 0x40130312:0x3ffeab50 0x400d5ccd:0x3ffeabb0 0x4012d477:0x3ffead60 0x4012ebf5:0x3ffeada0 0x4012ecd1:0x3ffeae30 0x4012e45c:0x3ffeae50 0x4012dcbe:0x3ffeae70 0x4012dd30:0x3ffeaeb0 0x40092ca1:0x3ffeaed0
> 0x4008734d
> i2s_intr_handler_default
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/driver/i2s.c:1258
> 0x400854a5
> _xt_medint2
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/freertos/xtensa_vectors.S:1243
> 0x4013202c
> rnet_lite_q_with_score_verify
> /home/gongxiaochao/code/esp/esp_temp/dl_lib/components/face_detection/mtmn/mtmn_lite_q.c:267
> 0x4012f86e
> rnet_forward
> C:/msys32/home/tatsutaigu/esp/esp-who/components/esp-face/face_detection/fd_forward.c:457
> 0x40130312
> face_detect
> C:/msys32/home/tatsutaigu/esp/esp-who/components/esp-face/face_detection/fd_forward.c:657
> 0x400d5ccd
> facenet_stream_handler
> C:/msys32/home/tatsutaigu/esp/esp-who/examples/single_chip/recognition_solution/main/app_httpserver.c:236
> 0x4012d477
> httpd_uri
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/esp_http_server/src/httpd_uri.c:218
> 0x4012ebf5
> httpd_parse_req
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/esp_http_server/src/httpd_parse.c:527
> 0x4012ecd1
> httpd_req_new
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/esp_http_server/src/httpd_parse.c:597
> 0x4012e45c
> httpd_sess_process
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/esp_http_server/src/httpd_sess.c:326
> 0x4012dcbe
> httpd_server
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/esp_http_server/src/httpd_main.c:113
> 0x4012dd30
> httpd_thread
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/esp_http_server/src/httpd_main.c:113
> 0x40092ca1
> vPortTaskWrapper
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/freertos/port.c:435

**Type 2**
> I (38263) app_httpserver: Get count 1
> 
> I (46133) face_recognition:
> Similarity: 0.156323, id: -1
> I (46723) face_recognition: ne_ratio  : 2.536366
> I (47323) face_recognition: ne_ratio  : 2.680747
> I (48093) face_recognition: ne_ratio  : 2.224622
> I (48983) face_recognition:
> Similarity: 0.051525, id: -1
> I (49883) face_recognition:
> Similarity: 0.056088, id: -1
> I (50803) face_recognition:
> Similarity: 0.042877, id: -1
> E (50803) to_bmp: JPG encoder init failed
> E (50803) app_httpserver: fmt2jpg failed

XiaochaoGONG, 2019-06-05:
About type 1 error, please do `make menuconfig` or `idf.py menuconfig` to increase the cpu interrupt watchdog timeout limit.
And did you update the submodules esp-face and esp32-camera ?

XiaochaoGONG, 2019-06-05:
About type 1 error, please increase the cpu interrupt watchdog timeout limit by `idf.py menuconfig` or `make menuconfig`

tatsutaigu, 2019-06-05:
I will try to increase watchdog timeout limit.

And, Type 2 is serious.
I got same error every time when it detects face.

BTW, I re-install esp-who Jun/5/2019 0:00 UTC.

XiaochaoGONG, 2019-06-05:
Please provide the commit of esp-who, esp-face, and esp32-camera.

tatsutaigu, 2019-06-05:
If I misunderstand, please let me know.

**esp-who**
> tatsutaigu@ABCDE-0123 MINGW32 ~/esp/esp-who
> $ git log -3
> commit 413bc90443c03e53add50ce3691f0830483d671f (HEAD -> master, origin/master, origin/HEAD)
> Author: XiaochaoGONG <gongxiaochao@espressif.com>
> Date:   Tue Jun 4 15:41:16 2019 +0800
> 
>     Update connection document.
> 
> commit ddd49bc617ec0e5afaaaedfc4c7aeacf95fd3b7a
> Author: XiaochaoGONG <gongxiaochao@espressif.com>
> Date:   Tue Jun 4 15:34:25 2019 +0800
> 
>     update docu
> 
> commit 74e076e152315b2fa5284f390df5bb87a945b8f1 (tag: v0.9.2)
> Author: XiaochaoGONG <gongxiaochao@espressif.com>
> Date:   Mon Jun 3 20:55:55 2019 +0800
> 
>     2672de6 esp-face mtmn quantization

**esp-face**
> tatsutaigu@ABCDE-0123 MINGW32 ~/esp/esp-who/components/esp-face
> $ git log -3
> commit 2672de6cda2e65686d87604fb81e8382a390eb9f (HEAD, tag: v0.4.1, origin/master, origin/HEAD, master)
> Merge: e69b536 cb70468
> Author: XiaochaoGONG <gongxiaochao@espressif.com>
> Date:   Mon Jun 3 20:46:57 2019 +0800
> 
>     Merge branch 'feature/mtmn_quant' into 'master'
> 
>     Feature/mtmn quant
> 
>     See merge request face-recognition-framework/esp-face!30
> 
> commit cb70468b73183216eb2231ffdf37b8d732b62814
> Author: XiaochaoGONG <gongxiaochao@espressif.com>
> Date:   Mon Jun 3 20:46:57 2019 +0800
> 
>     Feature/mtmn quant
> 
> commit e69b53600c8a066c5da85935972e572ee2426daf
> Author: XiaochaoGONG <gongxiaochao@espressif.com>
> Date:   Mon Jun 3 17:48:03 2019 +0800
> 
>     add component.mk for make system

**esp-camera**
> tatsutaigu@ABCDE-0123 MINGW32 ~/esp/esp-who/components/esp32-camera
> $ git log -3
> commit bd617234a74fa291e21dae6996a629af7f6ea5e5 (HEAD, origin/master, origin/HEAD, master)
> Author: me-no-dev <hristo@espressif.com>
> Date:   Sat Apr 13 22:38:53 2019 +0200
> 
>     Fix OV2640 not being reset when OV3660 is enabled
> 
> commit 67cf2403c01cfbecc0ea6ca7c662903cc2015384
> Author: Me No Dev <me-no-dev@users.noreply.github.com>
> Date:   Tue Apr 9 15:01:22 2019 +0200
> 
>     Update README.md
> 
> commit 161be67fe09db851fdbe2f00844b5049e8570ca9
> Merge: 113629b 57303f7
> Author: Me No Dev <me-no-dev@users.noreply.github.com>
> Date:   Tue Apr 9 14:59:55 2019 +0200
> 
>     Merge pull request #30 from espressif/ov3660-support
> 
>     Ov3660 support

XiaochaoGONG, 2019-06-05:
The version is right, so can you provide your sdkconfig file, and your esp-idf version ?

tatsutaigu, 2019-06-06:
**sdkconfig**
I changed CONFIG_INT_WDT_TIMEOUT_MS=800 to 1000 because I follow your advice for type 1 error.
[https://github.com/tatsutaigu/sdkconfig](https://github.com/tatsutaigu/sdkconfig)

**esp-idf**
> tatsutaigu@ABCDE-0123 MINGW32 ~/esp/esp-who/esp-idf
> $ printenv IDF_PATH
> /home/tatsutaigu/esp/esp-who/esp-idf
> 
> 
> tatsutaigu@ABCDE-0123 MINGW32 ~/esp/esp-who/esp-idf
> $ git log -3
> 
> commit 286202caa31b61c2182209f37f8069a0b60fb942 (HEAD, tag: v3.2-rc, tag: v3.2)
> Merge: f49e2afe9 9ebaf36c2
> Author: Ivan Grokhotkov <ivan@espressif.com>
> Date:   Mon Mar 25 10:49:56 2019 +0800
> 
>     Merge branch 'bugfix/mqtt_fragmented_packed_msg_3.2' into 'release/v3.2'
> 
>     MQTT support for fragmented and packed messages (Backport 3.2)
> 
>     See merge request idf/esp-idf!4593
> 
> commit 9ebaf36c288b325a0ab52058ce7381d030585089
> Author: David Cermak <cermak@espressif.com>
> Date:   Tue Nov 20 03:31:19 2018 +0100
> 
>     tcp_transport: modification of ws to read headers first in order to read the exact payload
> 
>     closes https://github.com/espressif/esp-mqtt/issues/69
> 
> commit 1ece141c52132aacdaf6c56d680ef91678a70dc6
> Author: David Cermak <cermak@espressif.com>
> Date:   Fri Nov 16 10:42:58 2018 +0000
> 
>     mqtt: support for BEFORE_CONNECT event in idf
> 
>     Updated examples to use new event id, idf to use mqtt with fixed retained, oversized messages
>

XiaochaoGONG, 2019-06-06:
Hi, I've compared the sdkconfig files, and found that `WIFI_STATIC_TX_BUFFER` should be set. In your sdkconfig, it's set as `Dynamic`. I tested with that and the same error occured. So you could change the wifi tx buffer to static mode, in menuconfig.

tatsutaigu, 2019-06-06:
Thank you for confirmation.

I change `DYNAMIC` to `STATIC`, then type 2 error disappeared.
I did not modify `WIFI_STATIC_TX_BUFFER` when menuconfig.
I guess menuconfig set automatically `DYNAMIC`.
`sdkconfig.default` may needs
```
CONFIG_ESP32_WIFI_STATIC_TX_BUFFER=y
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER=
```

BTW, I set CONFIG_INT_WDT_TIMEOUT_MS=2000.
However, type 1 error occurs.
If you have recommend value, please let me know.

XiaochaoGONG, 2019-06-06:
> BTW, I set CONFIG_INT_WDT_TIMEOUT_MS=2000.
I used your sdkconfig, and have no errors with that.

tatsutaigu, 2019-06-06:
Thank you for checking.

closed 2019-06-06 by tatsutaigu