espressif/esp-who #76: ESP-WHO Recognition_Solution - Not Working (ESP-EYE saying Hello ID 0 for Enrolled face and Not Enrolled face)

Yuvaraj4041, 2019-06-06:
Hi

I am trying "recognition_solution" Example under esp-who framework(Newly cloned).

After flashing(RED LED turned ON, Whit LED-OFF) the code, i am saying HI LEXIN (RED LED is Flashing, Whit LED-OFF).

Then, i am connecting with ESP-EYE Network and opened the ip 192.168.4.1/face_stream(White LED turned ON, Red LED-OFF).
LOG: app_wifi: station:20:e3:17:06:0d:8d join, AID=1[0m
tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2

After Single pressing Side Tactile Button it start enrolling the face and taking 5 Samples.
LOG : app_httpserver: Get count 0
app_httpserver: Enrolled Face ID: 0

Once it is done, i am showing the same Enrolled face and ESP Saying "Hello ID 0".
(Red LED is flashing)
LOG : face_recognition:
Similarity: 0.998691, id: 0

After i am Showing not enrolled face, then also it is saying "Hello ID 0".
(Red LED is flashing)
LOG : face_recognition:
Similarity: 0.996930, id: 0

Then i deleted all the enrolled face id's by double pressign Side Tactile button.
LOG : app_httpserver: 0 ID Left

Then again i started face enrolling, it is showing yellow box on the face, after within 2 second video streaming in the local ip got hanged and RED LED Switched ON, White goes OFF.

In This condition i am getting Error in Serial monitor :
LOG : to_bmp: JPG encoder init failed [0m
app_httpserver: fmt2jpg failed [0m

Some time i am getting this error after restarting the ESP-EYE, First time enrolling the face also.

XiaochaoGONG, 2019-06-06:
Duplicate of #71

closed 2019-06-06 by XiaochaoGONG

Yuvaraj4041, 2019-06-06:
Hi @XiaochaoGONG 

Thanks