espressif/esp-who #77: Documentation?

ChrisXenon, 2019-06-19:
I am struggling with ESP32-CAM on Arduino IDE> Not new to programming but new to ESP32 and camera interfaces.

Today's mystery - where is the defintion of dl_matrix3du_alloc()?
Clearly, from context in example code, it's allocating memory, presumably for the frame buffer. But why not just use the frame buffer, which is presumably, already in memory?

Obvioulsy there has to be a good reason to do the dl_matrix3du_alloc() - but I don't know what it is.

More generally - is there a place where all these functions are documented? I don't mean just example uses, I mean real documentation with text describing what each function does, and what each parameter is?

Thanks

me-no-dev, 2019-06-19:
Things have changed. I am about to update Arduino and the libs included inside :)

ChrisXenon, 2019-06-19:
Thanks. Will those changes answer my question about dl_matrix3du_alloc() or will they remove the need for me to know that answer?

Also, what do I do to receive these updates? Do I visit library manager often and check for updates there?

Thanks

ChrisXenon, 2019-06-20:
Perhaps you are extremelyt busy but I am having to invest a great deal of time here. I am trying to develope a commercial product and it's far from easy.

There seems to be a pattern to support - you respond rapidly to my first contact
but that response is inadequate and no further responses are made.

Liek above - you tell me changes are imminent - but when? And what? Will they render my code instantly non-functional? Should I stop unti lyou're done? Or what?

This is very frustrating. I am taking DAYS to find out by ugly trial-and-error what you could tell me in seconds. And if the documentation was there we would never have needed to speak at all.

What must I do to get the most basic of questions like "What does this function do?" answered?

me-no-dev, 2019-06-20:
Changes only mean that the Arduino lib will become in sync with this one. Whatever you see here, will be also in Arduino. TBH I advise you to not use Arduino for camera application development, but rather use IDF and this repo as submodule. Thing here move much more rapidly than in Arduino. There you need to wait for release to get the latest code.

Also I do get your frustration, but honestly I take care of the camera driver itself and not the ESP-FACE part, which is what you were asking about.

ChrisXenon, 2019-06-20:
Thanks for finally replying. Regaridng the Arduino IDE - I was under the impression that any code I copy from your Github will run on the Arduino IDE and this has been my experience so far. Is that wrong? If so, then I need to take your advice and get off it.

But - remarkably - you still haven't said where I find the documentation of definition of dl_matrix3du_alloc(). and my other thread remains outstanding. Whether it's you or someone else who should be doping it - they aren't doing it!

me-no-dev, 2019-06-20:
In the case of camera, you will need to drop the whole component, but at the same time it's already included in prebuilt form, which will probably cause a whole lot of issues. For general IDF code, it is true that you can compile it just fine without anything else. Still need to keep in mind that only a particular version of IDF is included in Arduino so you are limited to whatever it offers.

As for the other issue. I will advise you to show some patience. We all work on many many things and this is just one of them. You posted your issue yesterday... you need to give it some time.

Maybe check the esp-face repository (submoduled here) for source of that function.

ChrisXenon, 2019-06-20:
Thanks for your reply. Sorry but I do not know what "you will need to drop the whole component, but at the same time it's already included in prebuilt form, which will probably cause a whole lot of issues" means. What is "the camera component"? Drop it from what? Included where? Maybe I need to read some introductory stuff to establish a context from which you rcomments are meaningful? I don't want to waste your time with newbie stuff - I'd like to de-newbie on my own time, pawing over qualtiy documentation, but I cannot find it.

Regarding patience, I have been waiting since June 12th for a response on this:
https://esp32.com/viewtopic.php?f=19&t=10405&sid=e2107a7186f2f0b2421e0e1f63c07188
And I tihnk you'd have to agree that your first response on this conversation was not hepful in solving my problem.

I appreicate that you may be busy elsewhere. My comments should be seen in the context of criticising the service not the individual. If I wait for an hour to order a drink in a restaurant, that's not good enough. It may be that the only waiter is rushed off his feet and paid a pittance - but the service is still not good enough - though it's not his personal fault.

Before I made my first ever post I searched Google for dl_matrix3du_alloc(); I hate asking for help and will work hard to avoid it. But I found only usages - not the definition. The fact that you are suggesting I search around for it seems to support my obserevation that the documentation is lacking.

But I don't want to antagonise you. Firstly, it probably isn't your fault, and secondly I need you (hehe) and thirdly Espressif has changed tghe world with its products and I really appreciate it.

I now have the greyscale bitmap I wanted from the camera. I just chopped out the dl_matrix3du_alloc crud and a bunch of other stuff I just didn't understand and things got easier. I'm still uneasy about why it was there in the first place and frustrated that I couldnb't find out, and I will need to put it back soon because I do need video streaming, but anyway, I have my bitmap.

The problem now is that I need a focal point less than a cm from the lens and I can't get it to focus. I should have checked this before I started coding - but I may find a way around it. (this last part is just waffle - by all means ignore it).

Thanks again for your support.

me-no-dev, 2019-06-20:
Alright :) focus? Most OV2640s come with the lenses glued :( so focus adjustment is not easy. If yours is not glued, then you can refocus it to whatever you want. AMF I have a "microscope" here with naked OV2640 (just the sensor, no lenses) and I have a few OV3660s that can be refocused, so that is definitely possible. The dl_matrix things are for face detection and face recognition, which you probably do not need. If you want a basic code for the camera without any face detection stuff, maybe the shortest one is the readme of esp32-camera https://github.com/espressif/esp32-camera/blob/master/README.md has some examples that might shed some light :)
I don't visit esp32.com unless a colleague notifies me of something that I need to answer :) sorry for not following up there.

I had a quick search for `dl_matrix3du_alloc` and it seems that it's provided not as source, but as a blob and this is just it's header. This is how all IDF is provided to Arduino as well. It's all libs `libSOMETHING.a` and their headers. This is how the esp32-camera and esp-face are provided as well. This works fine if you do not find any issues and all cases are covered for you, but if you need to fiddle with the driver/code, you need to build from source, which in Arduino as-is is not that easy.

ChrisXenon, 2019-06-20:
Thanks again. I have two and both were glued. One popped lose easily but the other fought until the end.
The datasheet says the focus range is from 15cm and I can confirm that it won't focus any closer. Moving the lens barrel from fully closed to falling out won't fix it. So I'll need another lens.

Thanks for the info on the mystery function. I can rest easy leaving it deleted  as I don't need face recognition. I only wanted to see the definition in order to see what it does. It's same isn't super helpful. 
 Thanks also for the link; I found it last wek and it was somewhat helpful.

me-no-dev, 2019-06-20:
I have a few different styles of lenses so there are options out there. You can get any OV2640/OV3660 sensor with decent lenses and use it. The connector is the same :) I can not guarantee the distance you want. Less than a cm is not much :) what are you working on that requires that close distance? Maybe there are other ways to achieve the same effect?

ChrisXenon, 2019-06-21:
I am building a light spectrometer product. I already sell one which uses a
640 x 480 USB webcam as the sensor.
The diffraction grating spreads light into its colours and I need to get
all of the spectrum onto the sensor.
The lens needs to focus close to the plane of the grating. I don't know of
another optical path that would work, but I'm exploring options.
The huge benefit of this module is that it includes an HD camera, a
competent computing platform and a WiFi conenction all for £6.
If I have to go to conventional platforms (RPi, for example) and USB
webcams the price goes up massively.
Anyway - not your problem :o)

me-no-dev, 2019-06-21:
Check what I found: https://www.seeedstudio.com/OV2640-Fisheye-Camera-p-4048.html
This will plug right into your board and you can change the lenses to whatever you want :)

ChrisXenon, 2019-06-21:
Many Thanks! This one is pre-order only but there are others online which
can be ordered.
The next hurdle (again entirely my problem and not yours)  is embarrassing
for me to admit.
Previously the cameras I have worked with had IR filters in the back of the
lens barrel, and these were easily removed with a needle point.
I need them gone so they don't affect what the spectrometer measures.
I assumed the OV2640 would be the same but it isn't. I discovered only
yesterday that the glass IR filter seems stuck to the sensor itself and I
cannot remove it with a needle, though I've discovered I can damage it
quite nicely.

Dammit this is rudimentary priority stuff which I didn't do. All the work
on the code before even considering the optics and the IR filter! Moron.
It looks like the lens problem is solvable with a product like the one you
found, though I might need to compensate for lens distortion in the
software,
and I've found something online suggesting a seller (uctronics.com
<sales@uctronics.com>) may be able to provide the camera without the IR
filter fitted.
So I wrotw to them, but I have an ugly feeling I'd need to buy thousands of
unitd before they'd do a batch without the filter.
I'm thinking maybe I should be stacking supermarket shelves for a living.

Meh. Enjoy your weekend.

me-no-dev, 2019-06-21:
I have a couple of modules like the one in the picture. I believe they do not have the IR filter on the sensor (colors are more violet when I use it) and on top of that has a standard mount for lenses, so you can easily find one that will do. Mounting is different, but also standard (for FPGAs usually) and there are many options that use the same pinout
[image]

me-no-dev, 2019-06-21:
[image]

ChrisXenon, 2019-06-21:
Ahhh! That is very reassuring! Light at the end of the tunnel (IR light).
Thanks a lot - I really appreciate your help.

Cheers
Chris

closed 2022-09-21 by Auroragan