espressif/esp-who #80: delete_face_id_in_flash_with_name

peterGruen, 2019-06-26:
When I delete the last Name of the st_face_list, all names will be broken.

Every Face ID has the Entry Matched Face: ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒o▒?▒o▒?
The Error occurs only at the last Entry, not on the first, second, ..

I use this Code: 

int8_t left = delete_face_id_in_flash_with_name(&st_face_list, face_name);
					if (left > 0){
					ESP_LOGE(TAG, "%s ID Deleted", face_name);
					}
Peter

XiaochaoGONG, 2019-06-27:
Hi what's the length of the last Name in the `st_face_list` ? There is a limit that the size should be no more than `ENROLL_NAME_LEN`:
https://github.com/espressif/esp-face/blob/bb800f9abc2e336d585abf2d13035bdc5712f71d/face_recognition/include/fr_forward.h#L31

peterGruen, 2019-06-28:
When I try this: 
`sprintf(add_face, "listface:%s", head->id_name);`
`ESP_LOGI(TAG, " Face-ID: %s Lenght: %d \n",add_face,strlen(add_face));`
I get following output:
E (131774) app_delete:  Face-ID: listface:Face0 Lenght: 14
E (131774) app_delete:  Face-ID: listface:Face1 Lenght: 14

peterGruen, 2019-06-28:
When I try do delete the first Entry, I get:
 Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
But the Face ID is deleted and the names are ok. Deleting the second or third Face ID are ok, only the first and the last.

XiaochaoGONG, 2019-06-29:
Hi Peter, I have tested with my code, there was no problem. Can you post on some of your test code ?