espressif/esp-who #81: unstable camera_web_server demo

tatsutaigu, 2019-06-28:
Hi, 
I got below error on "camera_web_server" sample.
I use ESP32-WROVER-B + OV2640.
"Camera Pins" on menuconfig is "WROVER-KIT With OV2640 Module".

> I (67507) camera_httpd: MJPG: 5915B 13ms (76.9fps), AVG: 51ms (19.6fps), 0+0+0+0=0 0
> I (67547) camera_httpd: MJPG: 6001B 45ms (22.2fps), AVG: 47ms (21.3fps), 0+0+0+0=0 0
> I (67587) camera_httpd: MJPG: 6466B 32ms (31.2fps), AVG: 49ms (20.4fps), 0+0+0+0=0 0
> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
> Core 0 register dump:
> PC      : 0x4000814d  PS      : 0x00060031  A0      : 0x800ecd85  A1      : 0x3ffb05d0
> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000004  A5      : 0x00000000
> A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d15c9  A9      : 0x3ffb05b0
> A10     : 0x3f4018ac  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x3ffddf10
> A14     : 0x00000000  A15     : 0x3ffbd640  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
> Core 0 was running in ISR context:
> EPC1    : 0x4000814d  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000
> 
> Backtrace: 0x4000814d:0x3ffb05d0 0x400ecd82:0x3ffb05f0 0x400853dd:0x3ffb0610 0x401555af:0x3ffbef60 0x400d1b56:0x3ffbef80 0x40092099:0x3ffbefa0 0x40092add:0x3ffbefc0
> 
> Rebooting...
> ets Jun  8 2016 00:22:57

backtrace is here.
> $ xtensa-esp32-elf-addr2line -e ./build/camera_web_server.elf -af 0x4000814d:0x3ffb05d0 0x400ecd82:0x3ffb05f0 0x400853dd:0x3ffb0610 0x401555af:0x3ffbef60 0x400d1b56:0x3ffbef80 0x40092099:0x3ffbefa0 0x40092add:0x3ffbefc0
> 0x4000814d
> ??
> ??:0
> 0x400ecd82
> rtc_isr
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/driver/rtc_module.c:1944
> 0x400853dd
> _xt_lowint1
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/freertos/xtensa_vectors.S:1154
> 0x401555af
> esp_pm_impl_waiti
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/esp32/pm_esp32.c:487
> 0x400d1b56
> esp_vApplicationIdleHook
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/esp32/freertos_hooks.c:86
> 0x40092099
> prvIdleTask
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/freertos/tasks.c:3537
> 0x40092add
> vPortTaskWrapper
> C:/msys32/home/tatsutaigu/esp/esp-who/esp-idf/components/freertos/port.c:435
> 

My esp-who commit is here.

> $ git log -2
> commit 258751b5b615ffdc2e0f43f8815b6ffbaeb3da03 (HEAD -> master, origin/master, origin/HEAD)
> Author: Me No Dev <me-no-dev@users.noreply.github.com>
> Date:   Thu Jun 20 04:43:46 2019 +0200
> 
>     Add Save button to capture window (#78)
> 
>     * Add "Save" button to capture window
> 
>     Allows saving images from both stills and streams by clicking a button.
> 
>     * Make image filenames be the date and time of capture
> 
>     * Add script to compress the html pages for embedding
> 
> commit 7426e27e36be7b5a9de690e01e231f505da637ee
> Author: XiaochaoGONG <gongxiaochao@espressif.com>
> Date:   Wed Jun 19 12:25:06 2019 +0800
> 
>     bb800f9a update esp-face add frmn models
>

me-no-dev, 2019-06-28:
hmmm... power all good? The error does not point to any camera code and it's quite strange. Are you running the example as-is? I have never been able to crash it :)

tatsutaigu, 2019-07-18:
Hi @me-no-dev ,
root cause may camera connection.
connection was unstable, so, this error might occur.

closed 2019-07-18 by tatsutaigu

me-no-dev, 2019-07-18:
@tatsutaigu thanks for reporting back :) will keep this in mind if I see such issue again