espressif/esp-who #82: PlatformIO support.

Mnark, 2019-07-02:
I have successfully managed to use PlatformIO to build and run the web server example in esp-who. 

The changes require an additional file (library.json) to be added the root of each of the component projects (esp-face, esp-camera etc.) to enable those components to be installed into PlatformIO, as well as additions to the main esp-who repository (platform.ini).

So questions...
  Would you like these changes contributed to the repository via a pull-request? If so, should I submit a separate request for each component?
or
  Should i just post some detailed instructions here?

XiaochaoGONG, 2019-07-03:
Hi Mnark, thanks a lot ! This will help the user in #44 .
It's better you post some detailed instructions here :)

TjeerdvanDijk, 2022-01-14:
Hey, this would still be very useful. I think there are many people that can benefit from this.

zeljkoavramovic, 2022-08-29:
I would prefer pull request but instructions are also fine. Thanks!

JoseBarreto1, 2023-07-24:
I would also like instructions, to use with PlatformIO