espressif/esp-who #83: app_process: Camera capture failed

mctosima, 2019-07-08:
Anyone can help with app_process: Camera capture failed while monitoring ?
I'm using ESP-eye and following the instruction given on this link (https://github.com/espressif/esp-who)

The terminal shows 
E (97) camera: Detected camera not supported.
E (97) camera: Camera probe failed with error 0x20004
E (97) app_camera: Camera init failed with error 0x20004
E (97) app_process: Camera capture failed
E (107) app_process: Camera capture failed
E (107) app_process: Camera capture failed
E (107) app_process: Camera capture failed
E (117) app_process: Camera capture failed
.
.

XiaochaoGONG, 2019-07-08:
Did you open PSRAM in menuconfig ?

XiaochaoGONG, 2019-07-08:
Hi, did you enable SPI RAM in menuconfig ?

mctosima, 2019-07-09:
> Hi, did you enable SPI RAM in menuconfig ?

Thankyou Xiaochao. Problem solved by enabling SPI RAM (Component Config > ESP32 Specific > Support for External SPI RAM).
I'm little bit new in ESP.

[image]

closed 2019-07-09 by mctosima

Kirankumar-xenlabs, 2019-09-16:
change the frame size try it