espressif/esp-who #85: ESP-EYE sending image to AWS

mctosima, 2019-07-11:
Hi, just need little help to start my project. Are there any documentation to start learning ESP-EYE in sending images to Amazon Web Service ?

A few links or webpage should help me, thanks a lot.
Cheers

XiaochaoGONG, 2019-07-11:
Hi, this repo may help you. https://github.com/espressif/esp-aws-iot

chegewara, 2019-07-11:
This is based on esp_http_client and AWS Gateway:
(its not exactly meant to be used with esp-who, because i used it over a year ago with this repo https://github.com/chegewara/ESP32_CAMERA_QR, but you have some insight how it can be done)
```
/* ESP HTTP Client Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "app_wifi.h"

#include "esp_http_client.h"

#define MAX_HTTP_RECV_BUFFER 512
static const char *TAG = "HTTP_CLIENT";

extern const char server_root_cert_pem_start[] asm("_binary_server_root_cert_pem_start");
extern const char server_root_cert_pem_end[]   asm("_binary_server_root_cert_pem_end");

esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
	switch (evt->event_id) {
	case HTTP_EVENT_ERROR:
		ESP_LOGD(TAG, "HTTP_EVENT_ERROR");
		break;
	case HTTP_EVENT_ON_CONNECTED:
		ESP_LOGD(TAG, "HTTP_EVENT_ON_CONNECTED");
		break;
	case HTTP_EVENT_HEADER_SENT:
		ESP_LOGD(TAG, "HTTP_EVENT_HEADER_SENT");
		break;
	case HTTP_EVENT_ON_HEADER:
		ESP_LOGD(TAG, "HTTP_EVENT_ON_HEADER, key=%s, value=%s", evt->header_key, evt->header_value);
		break;
	case HTTP_EVENT_ON_DATA:
		ESP_LOGD(TAG, "HTTP_EVENT_ON_DATA, len=%d", evt->data_len);
		
		if (!esp_http_client_is_chunked_response(evt->client)) {
			// Write out data
			printf("%.*s", evt->data_len, (char*)evt->data);
		}

		break;
	case HTTP_EVENT_ON_FINISH:
		ESP_LOGD(TAG, "HTTP_EVENT_ON_FINISH");
		break;
	case HTTP_EVENT_DISCONNECTED:
		ESP_LOGD(TAG, "HTTP_EVENT_DISCONNECTED");
		break;
	}
	return ESP_OK;
}

#include "mbedtls/base64.h"

void https(void* data) 
{
	http_buffer_t *fb_data = (http_buffer_t*) data;
	size_t size = fb_data->size;
	ESP_LOGI(TAG, "%d", size);
	const uint8_t* image = fb_data->data;

	unsigned char *buffer = (unsigned char*) heap_caps_malloc((size + 2 - ((size + 2) % 3)) / 3 * 4 +2, MALLOC_CAP_SPIRAM);
	//memset(buffer, 0, sizeof(buffer));
	if (buffer == NULL)
	{
		ESP_LOGE(TAG, "go away, you have no idea what you are doing!!!");
		return;
	}
	heap_caps_print_heap_info(MALLOC_CAP_SPIRAM);
	size_t buff_size = 0;
	int err64 = mbedtls_base64_encode(buffer, (size + 2 - ((size + 2) % 3)) / 3 * 4 +2, &buff_size, image, size);
	if (err64 != 0) {
		ESP_LOGE(TAG, "error base64 encoding, error %d, buff size: %d", err64, buff_size);  //(size + 2 - ((size + 2) % 3)) / 3 * 4);
		return;	
	}
	
	ESP_LOGI(TAG, "buffer size: %d", buff_size);
	
	esp_http_client_config_t config = {
		.url = "https://xxxxxxxxx-api.us-east-1.amazonaws.com/prod/upload_to_s3",
		.event_handler = _http_event_handler,
		.cert_pem = server_root_cert_pem_start,
	//	.timeout_ms = 20000,
	};
	esp_http_client_handle_t client = esp_http_client_init(&config);
	int _s = buff_size + 16;
	char *post_data = heap_caps_malloc(_s, MALLOC_CAP_SPIRAM);   //"{\"image\":\"\"}";
	memset(post_data, 0, _s);
	
	memcpy(post_data, "{\"image\":\"", 11);
	memcpy(post_data + 11, buffer, buff_size);
	memcpy(post_data + 11 + buff_size, "\"}", 3);
	
//	int __s = snprintf(post_data, _s, "%s%s%s", "{\"image\":\"", buffer,"\"}");

	esp_http_client_set_method(client, HTTP_METHOD_POST);
	esp_http_client_set_header(client, "Content-Type", "image/jpeg");
	ESP_LOGW(TAG, "post data size: %d", buff_size);
	esp_http_client_set_post_field(client, (const char*)buffer, buff_size);
	esp_err_t err = esp_http_client_perform(client);
	if (err == ESP_OK) {
		ESP_LOGI(TAG,
			"HTTP POST Status = %d, content_length = %d",
			esp_http_client_get_status_code(client),
			esp_http_client_get_content_length(client));
	}
	else {
		ESP_LOGE(TAG, "HTTP POST request failed: %s", esp_err_to_name(err));
	}

	esp_http_client_cleanup(client);
	heap_caps_free(buffer);
	heap_caps_free(post_data);
}
```

closed 2019-07-31 by mctosima