espressif/esp-who #86: Enrollment not works a the first time

AdiDDumke123, 2019-07-11:
* @param l                     face id list
     * @param aligned_face          An aligned face
     * @param enroll_confirm_times  Confirm times for each face id enrollment
     * @return -1                   Wrong input enroll_confirm_times
     * @return 0                    Enrollment finish
     * @return >=1                  **The left piece of aligned faces should be input**
     */
    int8_t enroll_face(face_id_list *l, dl_matrix3du_t *aligned_face);

i use the method "**enroll_face_with_name**". But there is no description. 

The Enrollment dont works at the first time. Firstly i got the Value "2" for return. 
But i dont know, what is meaning with that.

XiaochaoGONG, 2019-07-12:
The return value is the sample needed to be enrolled, by default 3 times form a face id.

AdiDDumke123, 2019-07-16:
is the recognition improved by the number of images when logging in?

XiaochaoGONG, 2019-07-16:
Theoretically say, yes, we average the face ids.

closed 2019-09-04 by XiaochaoGONG