espressif/esp-who #89: compile error: build asio

newstarqu, 2019-07-12:
CXX build/asio/asio/asio/src/asio.o
In file included from /opt/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdlib:75,                                                                                                             from /opt/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/ext/string_conversions.h:41,                                                                                            from /opt/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/bits/basic_string.h:6391,                                                                                               from /opt/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/string:52,                                                                                                              from /mnt/f/work/esp/esp32/esp-who/esp-idf/components/asio/asio/asio/include/asio/impl/error.ipp:19,                                                                                  from /mnt/f/work/esp/esp32/esp-who/esp-idf/components/asio/asio/asio/include/asio/impl/src.hpp:22,                                                                                    from /mnt/f/work/esp/esp32/esp-who/esp-idf/components/asio/asio/asio/src/asio.cpp:11:                                                                                /opt/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'                                                            void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)                                                                                                                                                    ^~~~~~~~~~~~~~~~~~                                                                                                                        /opt/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'                                                                 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)                                                                                                                                                                      ^~~~~~~~~~~~~                                                                                                                     /mnt/f/work/esp/esp32/esp-who/esp-idf/make/component_wrapper.mk:285: recipe for target 'asio/asio/src/asio.o' failed
make[1]: *** [asio/asio/src/asio.o] Error 1
/mnt/f/work/esp/esp32/esp-who/esp-idf/make/project.mk:530: recipe for target 'component-asio-build' failed
make: *** [component-asio-build] Error 2

XiaochaoGONG, 2019-07-12:
Did you get the submodules in esp-idf?

newstarqu, 2019-07-15:
> Did you get the submodules in esp-idf?

Yes, I did.
it can be success when I set IDF_PATH to my old idf folder，but it will be failed when set IDF_PATH to esp-who/esp-idf.

XiaochaoGONG, 2019-07-16:
I think the 'recursive' modules inside the esp-idf is not fetched.

newstarqu, 2019-07-17:
> I think the 'recursive' modules inside the esp-idf is not fetched.

I use "git clone --recursive https://github.com/espressif/esp-who.git " to get source code, and there's no error.

Shoud I check branch for esp-idf?  The branch of esp-idf is master and esp-who/esp-idf is "HEAD detached at 286202caa".

XiaochaoGONG, 2019-09-04:
Hi I'm not sure your IDF_PATH is pointed to which esp-idf, and, if your `git status` is clean.

closed 2019-09-16 by XiaochaoGONG