espressif/esp-who #90: ESP always crashing when attach an Interupt.

kami83, 2019-07-14:
Hi,

if i use the standard CameraWebserver Example and put a normal:
pinMode(BRDButton, INPUT_PULLUP);
 attachInterrupt(BRDButton, resetModule, FALLING);

in the Setup-Code. I always get a crash when running the ESP32:

15:35:15.940 -> WiFi connected
15:35:16.075 -> Starting web server on port: '80'
15:35:16.075 -> Starting stream server on port: '81'
15:35:16.075 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
15:35:16.075 -> Core 1 register dump:
15:35:16.075 -> PC      : 0x401589c7  PS      : 0x00060d30  A0      : 0x800d41d0  A1      : 0x3ffb1ea0  
15:35:16.075 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x40081180  A5      : 0x00000000  
15:35:16.075 -> A6      : 0x3ffc12e4  A7      : 0x00000000  A8      : 0x800ded8c  A9      : 0x3ffb1e50  
15:35:16.075 -> A10     : 0x00000105  A11     : 0x0000040e  A12     : 0x00000000  A13     : 0x00000000  
15:35:16.075 -> A14     : 0x40081180  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
15:35:16.075 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
15:35:16.075 -> 
15:35:16.075 -> Backtrace: 0x401589c7:0x3ffb1ea0 0x400d41cd:0x3ffb1ec0 0x400d4235:0x3ffb1ef0 0x400d330e:0x3ffb1f10 0x400d4d33:0x3ffb1fb0 0x40091e41:0x3ffb1fd0
15:35:16.075 -> 
15:35:16.075 -> Rebooting...
15:35:16.075 -> ets Jun  8 2016 00:22:57
15:35:16.075 -> 
15:35:16.075 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:35:16.075 -> configsip: 0, SPIWP:0xee
15:35:16.075 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:35:16.075 -> mode:DIO, clock div:1
15:35:16.075 -> load:0x3fff0018,len:4
15:35:16.075 -> load:0x3fff001c,len:1100
15:35:16.075 -> load:0x40078000,len:10088
15:35:16.075 -> load:0x40080400,len:6380
15:35:16.075 -> entry 0x400806a4

Can someone help me? I would like to react for a button press.

Thanks a lot.

Cu kami

me-no-dev, 2019-07-14:
which pin are you using for button? which camera board?

kami83, 2019-07-15:
Hi,

i defined Pin 12 as BRDButton and i am using the ESP32-CAM - A.I. Thinker Board.

Can you help me?

Cu kami

me-no-dev, 2019-07-15:
pin 12 should be OK. Since you are using Arduino, please decode the stack trace and post the source of `resetModule()`. Instructions on decoding the stack trace can be found in the ESP32 Arduino repository README :)

kami83, 2019-07-15:
Hi,

sorry for the late answer. Have to come home.

PC: 0x401589d7: esp_intr_get_cpu at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/intr_alloc.c line 780
EXCVADDR: 0x00000000

Decoding stack results
0x401589d7: esp_intr_get_cpu at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/intr_alloc.c line 780
0x4012f888: gpio_isr_handler_add at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/gpio.c line 373
0x400d7ec3: camera_init at /Users/ficeto/Desktop/ESP32/ESP32/public-libs-build/components/esp32-camera/driver/camera.c line 1044
0x400d802e: esp_camera_init at /Users/ficeto/Desktop/ESP32/ESP32/public-libs-build/components/esp32-camera/driver/camera.c line 1102
0x400d32ba: setup() at C:\Users\kami\Desktop\facescanner2/facescanner2.ino line 124
0x400d4d43: loopTask(void*) at C:\Users\kami\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\cores\esp32\main.cpp line 20
0x40091e71: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

void resetModule() {
 digitalWrite(13, HIGH);
}

resetmodule is just a simple function. It happens with every function.

Cu kami

me-no-dev, 2019-07-16:
I think that the problem is that both IDF's GPIO ISR and Arduino's ISR are used in your code and they overwrite each other. Try to not use interrupt and just check for the pin in a task or the loop. A valid press always lasts more than 100ms, so you should be OK to test it in the loop. Please try and let me know. I'll try to think of a way to make this work in the future

kami83, 2019-07-16:
Okay, thank you. I will try it at home.

But to attach an Interrupt would be great.

Cu kami

bigbrassbed, 2019-08-04:
your stacktrace shows that you fail at   intr_alloc.c line 780  with a null-ptr -> unhandled exception
Reason: before calling   "gpio_isr_handler_add"     we had called "gpio_install_isr_service". Here we forgot to check the outcome of the call and just continued. The outcome was ( on the way passing some other procedures) ESP_ERR_NOT_FOUND 0x105 triggered when attempting to find a slot for the interrupt related to the isr  -> no slot for this specification left :-)  Presumably you have already called before in your code  attachInterrupt( ...) e.g. to get some PIR working.  If you place the attachInterrupt behind the call to  esp_camera_init  then you got an unhandled exception caused by the attachInterrupt call. 
question to me-no-dev : why can't we allocate a another interrupt with the same spec - or are all interrupts already in use? We haven't basically done anything up to this step.

bigbrassbed, 2019-08-05:
so got it :-)
we can only once initialize the GPIO interrupt service - either via attachInterrupt or via esp_camera_init  (interrupt needed for vsync).
Since we use the camera, we can't use attachInterrupt with his low level initializing of GPIO interrupt service.
=> Replace call to attachInterrupt with something like (compare https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/gpio.html)
...
  err = gpio_isr_handler_add(GPIO_NUM_13, &detectsMovement, (void *) 13);
  if (err != ESP_OK) {
      Serial.printf("handler add failed with error  0x%x \r\n", err);
  }
  err = gpio_set_intr_type(GPIO_NUM_13, GPIO_INTR_POSEDGE);
  if (err != ESP_OK) {
      Serial.printf("set intr type failed with error  0x%x \r\n", err);
  }
...

and the handler func:
static void IRAM_ATTR detectsMovement(void * arg) {
  Serial.println("MOTION DETECTED!!!");
}

kami83, 2019-08-05:
Thanks a lot.

Looks really good. I will try it later on.

Cu kami

Adrianotiger, 2019-12-04:
> so got it :-)
> we can only once initialize the GPIO interrupt service - either via attachInterrupt or via esp_camera_init (interrupt needed for vsync).
> Since we use the camera, we can't use [...]

Thank you for this solution! I was able to add the interrupt now.

vks007, 2020-02-23:
I had the same issue and resolved it by the solution provided by @bigbrassbed . Thank you so much. 
The only extra thing to note is that the below coder should be called after the camera init else it will rightly fail with an isr initialization error
```
...
esp_err_t err = esp_camera_init(&config);
...

```> err = gpio_isr_handler_add(GPIO_NUM_13, &detectsMovement, (void *) 13);
> if (err != ESP_OK) {
> Serial.printf("handler add failed with error 0x%x \r\n", err);
> }
> err = gpio_set_intr_type(GPIO_NUM_13, GPIO_INTR_POSEDGE);
> if (err != ESP_OK) {
> Serial.printf("set intr type failed with error 0x%x \r\n", err);
> }
>... 

@kami83 , hope you were also able to get it, you should close this issue now.

closed 2020-02-24 by kami83

chandana9123, 2020-03-24:
Why interrupt is not working properly with esp32cam?? May i know the reason for this issue

here is my code
const byte ledPin = 14;
const byte interruptPin = 2;
volatile byte state = LOW;

void setup() {
  pinMode(ledPin, OUTPUT);
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), blink, CHANGE);
}

void loop() {
  digitalWrite(ledPin, state);
}

void blink() {
  state = !state;
}

me-no-dev, 2020-03-24:
@chandana9123 this question is not for this repository. it's for Arduino.

sayaboinajagadeeshwar, 2020-06-29:
Hey everyone,

I am also using Interrupts and Camera functionalities in my program in order to design a QR code scanner out of ESP32-CAM.
The interrupt button, when pressed once starts scanning the QR code but the process keeps on going without stopping. I have programmed the device such that it should scan only when I press the button attached with the interrupt pin. But this does not happen and I receive the error as stated in the below stack trace when I press the button attached to interrupt pin again.
I have decoded the stack trace below:-

Decoding stack results
0x40095f00: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x40096131: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x40104683: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x400873df: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tracecaps.c line 267
0x400878ad: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 42
0x400d5131: quirc_destroy at C:\Users\sayab\OneDrive\Documents\Arduino\libraries\ESPIno32CAM\src\quirc\quirc.c line 40
0x400d221f: qr_recoginze(void*) at C:\Users\sayab\OneDrive\Documents\Arduino\libraries\ESPIno32CAM\src\ESPino32CAM_QRCode.cpp line 97
0x40092649: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

jshep321, 2020-07-29:
much thanks @bigbrassbed.  Was banging my head here.

UllaBritta, 2023-03-21:
i was trying to fix this but ended up in another rebooting loop

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
Connecting to 2DEB-WiFi
.
WiFi connected.
IP address: 
192.168.99.183
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d2a17  PS      : 0x00060f30  A0      : 0x800d5e28  A1      : 0x3ffb27c0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3ffc3ea8  A5      : 0x00000004  
A6      : 0x3ffb8874  A7      : 0x80000001  A8      : 0x800d2a0c  A9      : 0x3ffb2790  
A10     : 0x00000031  A11     : 0x3f400150  A12     : 0x00000006  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x0000001a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400d50aa  LEND    : 0x400d50b6  LCOUNT  : 0x00000003  

Backtrace: 0x400d2a14:0x3ffb27c0 0x400d5e25:0x3ffb2820

ELF file SHA256: 462b770874885f62

help would be greatly appreciated :D