espressif/esp-who #95: Error Compiling fd_forward.h: No such file or directory

Aiouan, 2019-07-23:
Any ideas please?

Aiouan, 2019-07-24:
nvm I fixed it, I had downloaded the zip file which is incomplete.
If anyone else gets this or similar, I had to use:
git clone --recursive https://github.com/espressif/esp-who.git 
in the esp directory.

closed 2019-07-24 by Aiouan

npalazz, 2021-12-26:
hi! I have the same problem. I use an ESP32 camera OV2460 and when I compli I get the same error. I don't undertans what to do! :/ Can you give me a hand? Thank
[image]
s

ESP-YHY, 2022-03-07:
Hi @npalazz, please use the latest master branch of ESP-WHO.

drigocodeup, 2025-02-26:
Olá pessoal , estou com o mesmo problema, já baixei o repositório aqui no git para ide, já inclui no scket mas nada, já clonei no  git, porém parece que não vem o arquivo fd_forward.h, alguém tem esse arquivo?