espressif/esp-who #96: Add a button to enable LED flash on ESP32-CAM AI Thinker Boards

bkeevil, 2019-07-28:
I require the use of the LED on the AI Thinker ESP32-CAM board in the Web Cam Server demo and was thinking of preparing a patch to implement this feature request (See also #24). 

Any suggestions on the best way to do this?

bkeevil, 2019-07-28:
My use case is for a web cam that can do timelapse movies of a 3D printing process. The timelapse is done by Octoprint by requesting JPEG still images from the ESP32-CAM module via the web server demo app. I was thinking of adding a slider to the web page to enable the LED to "flash" when requesting a still picture.

bkeevil, 2019-07-28:
I'm working on this feature in the development branch of my fork of this project.

https://github.com/bkeevil/esp-who/commit/55b4f13401a66b8c92e164df866973e37a1eb592

bkeevil, 2019-08-05:
Pull request #99

closed 2019-08-05 by bkeevil

corbolais, 2020-10-09:
https://github.com/xenpac/ESP32-CAM-with-LED-control